dmroeder/pylogix #1: More than one call to .GetTagList() results in error

thatcadguy, 2016-11-16:
I get the following when making consecutive calls to GetTagList():

C:\Users\Jake\Downloads\pylogix-master>python
Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:40:30) [MSC v.1500 64 bit (
AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from eip import PLC
>>> test=PLC()
>>> test.IPAddress="192.168.45.92"
>>> test.GetTagList()
[<eip.LGXTag instance at 0x000000000248D808>, <eip.LGXTag instance at 0x00000000
0248DC48>, <eip.LGXTag instance at 0x000000000248DC08>, <eip.LGXTag instance at
0x000000000248DC88>, <eip.LGXTag instance at 0x000000000248DCC8>, <eip.LGXTag in
stance at 0x000000000248DD48>]
>>> test.GetTagList()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 114, in GetTagList
    return _getTagList(self)
  File "eip.py", line 292, in _getTagList
    forwardOpenFrame = _buildTagRequestPacket(self, partial=False)
  File "eip.py", line 611, in _buildTagRequestPacket
    request = _buildTagListRequest(self, partial)
  File "eip.py", line 890, in _buildTagListRequest
    TLRequest += pack('<BB', 0x24, self.Offset)
struct.error: ubyte format requires 0 <= number <= 255
>>>

thatcadguy, 2016-11-16:
Formatting got screwed up

```
C:\Users\Jake\Downloads\pylogix-master>python
Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:40:30) [MSC v.1500 64 bit (
AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from eip import PLC
>>> test=PLC()
>>> test.IPAddress="192.168.45.92"
>>> test.GetTagList()
[<eip.LGXTag instance at 0x000000000248D808>, <eip.LGXTag instance at 0x00000000
0248DC48>, <eip.LGXTag instance at 0x000000000248DC08>, <eip.LGXTag instance at
0x000000000248DC88>, <eip.LGXTag instance at 0x000000000248DCC8>, <eip.LGXTag in
stance at 0x000000000248DD48>]
>>> test.GetTagList()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 114, in GetTagList
    return _getTagList(self)
  File "eip.py", line 292, in _getTagList
    forwardOpenFrame = _buildTagRequestPacket(self, partial=False)
  File "eip.py", line 611, in _buildTagRequestPacket
    request = _buildTagListRequest(self, partial)
  File "eip.py", line 890, in _buildTagListRequest
    TLRequest += pack('<BB', 0x24, self.Offset)
struct.error: ubyte format requires 0 <= number <= 255
>>>
```

dmroeder, 2016-11-17:
Woops, I'll look into that one

closed 2016-11-18 by dmroeder

thatcadguy, 2016-11-18:
No problem, thanks for fixing. Cheers from the PLCS.net forum, I'm Epy there.

dmroeder, 2016-11-18:
Ahh, I figured you were from PLC talk, but I wasn't sure which user  I knew that avatar looked familiar!

dmroeder, 2016-11-24:
Since you were using GetTagList, I thought I'd mention that I updated it to also return program scoped tags