dmroeder/pylogix #2: Multi-Dimensional Arrays?

brendonparker, 2016-11-22:
First off, this is awesome! Big help to get me up and going.

Second, thanks for the rickroll.

Third, I'm trying to read a multi-dimensional array. Something like: `MyTag.Data_Storage[0,0]` but the code doesn't seem to parse the response very well. I'm digging in, but wondering if this is something you've had working and/or tested? 

Thanks!

dmroeder, 2016-11-22:
1) Thanks
2) I was wondering when someone would notice.  Sorry for my immaturity
3) I have tested multi dimension arrays and they work for me. It appears that the member of the UDT has 2 dimensions, Logix doesn't allow me to do that.  Could you send me the UDT you are using?  Also, what version of Logix?

brendonparker, 2016-11-22:
See attachment for the UDT and the version: Studio 5000 V27

dmroeder, 2016-11-22:
I must be terrible at this, I don't see an attachment.

brendonparker, 2016-11-22:
Try this: https://dl.dropboxusercontent.com/u/15576660/UDT_DataStorage.L5X

dmroeder, 2016-11-22:
Your tag doesn't appear to be multi-dimension.  I don't think Logix allows you to even create multi-dimensions as a UDT member.  In other words you cannot have a tag like you showed above (MyTag.Data_Storage[0,0]).  When you look in your tag database, do you see your tag exactly how you described?  If so I'd love to get a copy because I didn't think it was possible.  You can declare an instance of a UDT multi-dimensional but you can't have multi-dimensional members.

I see you have two members of your type.  DataStorage and DataStorage_Hold.  Are you wanting to get values from both members at the same time?  In other words you expecting to get "MyTag.DataStorage[0]" and "MyTag.DataStorage_Hold[0]" by attempting to read "MyTag.DataStorage[0,0]"?

brendonparker, 2016-11-22:
FWIW the tag I am trying to read is `BCS4201.Results[0,0]`

dmroeder, 2016-11-22:
Would it be possible to get a copy of the program?  If you don't feel comfortable sharing it, could I at least get a copy of it stripped down to just that tag?

dmroeder, 2016-11-22:
Is there something beyond .Result[0,0].  Maybe like .Result[0,0].DataStorage?

brendonparker, 2016-11-22:
Thanks for quick replies. I figured it out. It was user error on my part.
Although when I called two different array tags back to back I was getting
an error, so that threw me off. Haven't had time to dig into that yet. Will
let you know if that too is my error :P

dmroeder, 2016-11-22:
Yeah no problem, thanks for getting back.

dmroeder, 2016-11-24:
Since I didn't hear anything back, I'm closing the issue.

closed 2016-11-24 by dmroeder