dmroeder/pylogix #4: tags that are aliases

sla29970, 2017-02-10:
The example.py ex_getTags() correctly prints the datatype of tags that are aliases, but ex_read() and ex_write() fail on tags that are aliases because they find a datatype of 2 which is not one of the elements in self.CIPTypes

dmroeder, 2017-02-11:
The fact that you have an alias involved is not really relevant.  It's still a tag, it's just "linked" to another tag.  My suspicion is that the tag you are referring to might created as a program scoped tag.  Could you describe the tag you are trying to read, it's data type and the scope it was created in?  If I'm correct that it is a program scoped tag, what is the name of the program that it is created in?

dmroeder, 2017-02-11:
Wait, I think I know what is happening.  Can you successfully read or write the tag without performing the "GetTags" first?  So only do the "ex_read() and see if it's successful.

sla29970, 2017-02-11:
Yes.  The ability to read a tag works if GetTags is not done first. Writing to an alias tag always fails.

dmroeder, 2017-02-11:
I just pushed a commit that should fix the issue (commit 107).  See if that fixes your issue.  If it does not, please tell me the tag name, scope and data type you are trying to read/write.

sla29970, 2017-02-13:
Thanks.  Reading and writing alias tags now works with and without doing getTags.

dmroeder, 2017-02-13:
Great, thanks for the update.

closed 2017-02-13 by dmroeder