dmroeder/pylogix #5: GetTagList bug?

MarcosZornitta, 2017-03-17:
I found a little bug in the 'GetTagList' function ... the taglist is never emptied, so when  the function is called twice or more (for the same PLC), it adds the tags that are already in the list, again. 
So to fix this I had to access the global list and empty it every time the function is called.

Nothing too disturbing, but ...

dmroeder, 2017-03-17:
Good catch, I'll get this fixed tonight.  Thanks for reporting.

dmroeder, 2017-03-18:
I pushed out a fix for that one.

MarcosZornitta, 2017-03-20:
You need to empty the ProgramNames list as well, or the error will continue for all tags outside the scope of the controller.

`self.Offset = 0`
`del taglist[:]`
`del programNames[:]`

dmroeder, 2017-03-20:
You're right.  The program I tested didn't have program scoped tags so I missed it.  I'll get that fixed tonight.  Thanks.

closed 2017-03-21 by MarcosZornitta