dmroeder/pylogix #8: Micro820

shah-s, 2017-07-24:
Hey I am trying to read data from Micro820 PLC and tag i am trying to read is DV tag. 
Can your code be helpful to me?

dmroeder, 2017-07-24:
I have never worked with a Micro820, I only intended this project to work with CompactLogix and ControlLogix controllers.

However, unrelated to this project, I recently requested a Micro820 demo from Rockwell, I'm not sure when I will receive it but I may look into this when I get time.  No guarantees though...

closed 2017-07-24 by dmroeder

dmroeder, 2017-07-24:
I'm hearing that there is only a minor difference when handling the Micro800 line, I'll see what I can do to support it when I get my demo unit.

redmeters-sshah, 2017-07-24:
okay that will be a great help.
thank you :)

dmroeder, 2017-08-01:
I was read tags with some minor changes, though only global variables.  If you want to help test it, email me, otherwise it may take me a bit to do more testing.

redmeters-sshah, 2017-08-02:
yes I am also trying to read global read tag that is DV tag, if you can
help me with that.

mglowinski93, 2019-04-21:
Did you guys found a solution to read local variable on Micro820?