dmroeder/pylogix #9: ONE CLP, TWO PYLOGIX PROGRAMS

MarcosZornitta, 2017-09-08:
I need two different programs to connect to a PLC to pick up different information. But the CLP ethernet card shows the 'EXCP 0300' fault when I do this. Do I need to do something different for this case?

dmroeder, 2017-09-09:
You shouldn't have to do anything special to use multiple instances reading at the same time.  You just have to respect the modules limit of simultaneous connections.  Is this a 1756-ENBT?  If so, what firmware revision is it? 

There was an "undocumented feature" in specific firmware revisions that caused a memory leak apparently, which triggered this error message.  

If you can tell me what specific hardware you are communicating to and what firmware revisions they are, I may be able to help more.

MarcosZornitta, 2017-09-09:
Yes, it's a 1756-ENBT/A. The firmware is in revision 6.006.

The hardware structure is:

[image]

Note that I have two ethernet cards, but I only use slot 7 to communicate with pylogix. The other card is for automation devices only.

Thanks in advance.

dmroeder, 2017-09-09:
Can you give me a little context on how/when the this happens?  Can you share any of your code?  You can email directly to me if you prefer.

MarcosZornitta, 2017-09-09:
I still can not detect when this happens. However, if I leave only one program running the error does not happen (Only one program has been run for a month without problems). If I start reading tags in both programs, it reads without problems for a certain time and then displays the failure on the card. The codes do not have anything different, just read a bit (trigger) every 500ms and if it rises true the code starts reading about 1000 tags. Both programs do this, just change the tags that each reads.

But if you still need me to send you the code, I'll send it in your email.

dmroeder, 2017-09-12:
I have a ENBT and L73 at work that I will test with when I get some time.  I cannot reproduce the issue with a CompactLogix, I think this is more of a ENBT firmware issue. I'll do my best to prove it out though.  I'll make sure my ENBT has the same revision.  The ENBT did have a memory leak issue, though according to the KB, you are running the firmware that was supposed to fix it.  In fact, I think that is the latest.

http://www.plctalk.net/qanda/showthread.php?t=88929

I have been running a test over the last couple of days with my CompactLogix.  I have 4 scripts running, each reads ~750 tags every 20 seconds.  Each file has read ~6500 times so far with no issues

MarcosZornitta, 2017-09-12:
I was working with the ENBT in revision 6.004, but after searching in some forums and rockwell notes I updated to 6,006.

With some tests I realized that only one of my scripts is generating the problem, not the two together.
In this script I make connection to 7 PLCs (6 CompactLogix L18 and 1 ControlLogix L73).

For each connection I create a thread that every 1 second reads a bit (trigger) and after that it reads 6 other tags coming from a buffer that I created in the PLC.

Only in the case of L73, I have two Buffers in the same PLC (in my case it is a buffer for product entry in a freezing tunnel and another buffer for product output of this tunnel), so there are two threads in the same script that connect to the same PLC.

In each thread I work with Lock () before each reading to ensure that no attempt to perform simultaneous connection.

Can the error be in these two threads connecting to the L73?

Note: L18 PLCs never presented the problem, only the L73.

MarcosZornitta, 2017-09-13:
Can you send me your email to me? I'll send you two short scripts that when running both at the same time the card presents the EXCP 0300 error.

marcoszornitta@gmail.com

dmroeder, 2017-09-14:
I sent you an email, I'll check into it when you send the scripts over.

dmroeder, 2017-09-29:
The cause of the error on the ENBT seemed to mostly have to do with calling .Close() with every read/write. Close() should really only be called when you are completely done exchanging data with the controller. The reason being, there’s a bit of overhead when a connection is made. If you were to make a script that just did one simple read it would work like this:

-Register session
-Forward Open
-Initial Read to get data type
-Read to get the data
-Close() is called by the user
-Forward Close
-Unregister Session

Now if you had a scrip where a thread was continuously reading a single tag, you would have the register session, forward open and initial read happening one time, now you’d just be calling Read() since the connection is open and we now know the data type. If Close() is called, we’d have to re-register the session and reopen the connection. This would be very inefficient and in the case of the ENBT, it was too much for it causing the error on the card.

So the moral of the story is, we should only call .Close() when we are completely finished exchanging

closed 2017-09-29 by dmroeder