dmroeder/pylogix #10: multiParser on micro820 returns empty

limsan, 2017-10-22:
works on single tag, but multiTag still have problem
stripped = data[50:]   # stripped is empty but data [0:50} there are bytes

dmroeder, 2017-10-22:
Thanks for the info, I'll dig into this tomorrow when I have access to a Mirco820

dmroeder, 2017-10-23:
Turns out that the Micro8xx doesn't support the multi-read service.  I'll make some changes so it will fail more gracefully.

limsan, 2017-10-24:
Thank You, for clearing it up, a great project btw, I like your approach better, straight-forward & gets the job done rather than others similar, learned a lot from your code. thanks

closed 2017-10-24 by limsan

dmroeder, 2017-10-26:
I appreciate the kind words.  I tried to make it simple to use, mostly because I'm a simpleton!