dmroeder/pylogix #11: STRING32 Read/Write

AlineCco, 2017-11-10:
Hi, have any easier way to read/write user defined strings than char by char in DATA array?

I have two UDT Strings called 'STRING32' (with 32 max size) and 'STRING64' (with 64 max size).

dmroeder, 2017-11-10:
Unfortunately there is no "easy" way to write to a custom length string other than writing to .DATA[].  You can write an entire list to .DATA[] though like this:

thing = 'This is my string'
stuff = [ord(c) for c in thing]
comm.Write('CustomString.LEN', len(stuff))
comm.Write('CustomString.DATA[0]', stuff)

Custom  length strings appear as UDT's like any other UDT, so there is no easy way to write to them.  I am experimenting with reading/writing UDT's, but it still requires some work on the users end.  You'd have to know how the data is packed in the reply as well as how to pack it properly to send it.

dmroeder, 2017-11-14:
I gave you a solution that isn't quite right.  The problem with what I have above is if the original text is more characters than what you are writing, it will not overwrite them.  The proper solution is to write to write all 32 or 64 bytes of your string.  So something like this:

string_size = 32
text = 'This is some text'
stuff = [ord(c) for c in text] + [0] * (string_size - len(text))
comm.Write('CustomString.LEN', len(text))
comm.Write(CustomString.DATA[0]', stuff)

This will turn your string into a list, with 0's padded at the end.

velmurugan1, 2018-03-11:
Dear all i am using this pylogix code i have face one proble like when connected it automatically disconnect some time like 5 minute some time 2 minute

closed 2018-08-28 by dmroeder