dmroeder/pylogix #12: Error

Leobraic, 2018-01-30:
I have this message when I try to run the programme: **NameError: global name 'pack' is not defined**

What this mean?

dmroeder, 2018-01-30:
'pack' is part of the struct module. It's most likely with how struct is being imported.  You may need to provide a little more of the error in order to figure out what file is throwing the error but make sure that eip.py imports struct like this:

from struct import *

Leobraic, 2018-01-30:
I just put all of your .py files in a folder. In this same folder I put my .py file. I also create a __init.py__ in the folder. I have to modify any file? Or I just have to put all together and run?

dmroeder, 2018-01-31:
You don't have to modify any of my files to use them.  You just have to create an instance of PLC() properly, set the IP address (and processor slot if other than 0), then call Read or Write.  So I'd be interested in what is in your init.py (assuming that is where you are you are creating an instance of PLC).

Also, you should be using python 2.7.x, not python 3.  If you would like, I can take a look at how you are using it.  You can email it directly to me if you prefer.

Leobraic, 2018-01-31:
Undestand. I'm using python 2.7.x. My init.py is an empty file, I created this file because I read in a lot of websites that python needs an init.py to run. I don't know why, I just put it in my folder. 

Thank you for the answer, I'm going to test one more time tomorrow and then I reply this issue!

Have a good night!

dmroeder, 2018-01-31:
I see.

If you check out the readme, it has a simple example.   Create a new python file, you can call it read.py.  In it you would put something like this:

from eip import PLC
comm = PLC()
comm.IPAddress = '192.168.1.10' # replace with the IP address of your PLC
print comm.Read('TagName')
print comm.Read('TagName2')
comm.Close()

(You should only call .Close() when you are done reading/writing.  It doesn't need to be called after each read/write).

Let me know ether way how it goes.

Leobraic, 2018-01-31:
Yes. Thanks. I put exactly this code, but I didn't put comm.Close(). I think that wasn't the problem.

dmroeder, 2018-01-31:
Hmm, calling .Close() won't cause a problem.  It just closes the connection with the PLC.  If you don't call it, the connection would be dropped by the PLC eventually anyway.  

The only way I've really been able to re-create the same error is by modifying the import of struct in eip.py, which you shouldn't be doing.

If you want, zip up what you have and email it to me when you get a chance.  Also, mention your operating system and the PLC you are connecting to.

Leobraic, 2018-01-31:
The code works now. I just remove old .py files and I downloaded new ones. Thank you for support. I have one last question: Your code uses only the network connection to read or write data in an PLC or it uses an OPC server? Can I communicate with an PLC using Profinet, Modbus and etc?

dmroeder, 2018-01-31:
Great, thanks for the update. You can only communicate with ControlLogix, CompactLogix and Micro8xx, Ethernet I/P only.

closed 2018-01-31 by dmroeder

Leobraic, 2018-01-31:
Got it! Will you develop more communication protocols?

Piepiopi, 2018-01-31:
AB will only communicate over EtherNet I/P on those products. If you are coming from a different protocol, you will need a bridge/server to allow translation between the two protocols. Those PLCs are not going to respond to Profinet or ModBus TCP requests.

dmroeder, 2018-01-31:
I hadn't really planned on it Leobraic.  There are other libraries out there.  In fact, there are other Ethernet I/P libraries as well.  This has been a learning project for me.  I almost exclusively work with AB products so it wouldn't be very easy for me to work with other protocols.

Leobraic, 2018-01-31:
The question was because I'm new on PLC world. My first contact with PLC was in an AB PLC. Now I'm run to learn more and more information about PLCs but I'm having some difficulties.

Leobraic, 2018-01-31:
The function gets 2 arguments. But, if I request these two arguments from user. I have to pass:

**readArray('%s', variable_name, '%d', lenght_of_array**

And the programme give me an error: readArray() takes exactly 2 arguments (4 given).

It's impossible to do what I'm doing?

dmroeder, 2018-01-31:
The example is showing you how to read arrays, provide a tag name and
length and it will return a list of values.  Like this:

values = comm.Read('TagName', Length)

The function in the example, if you look at it, accepts 2 arguments.  tag
and length.  I'm not sure what your method of getting user input is (args
passed, gui, etc) but get the information from the user first, then pass it
to either the readArray example, or use Read directly.  Like this:

TagName = str(TagFromUser)
Length = int(LengthFromUser)

values = comm.Read(TagName, Length)

Leobraic, 2018-01-31:
Yeah thanks. The code works now. Thank you for all!