dmroeder/pylogix #13: Pylogix

velmurugan1, 2018-03-11:
Hai sir frequently its disconnected supposes i ask the data in while loop it work 30 sec and i ask every 1 sec it work 15  min i ask every 500 ms it WOR 6 min i can't find out pblm

dmroeder, 2018-03-11:
Are there any error messages you can share?
Are you calling .Close() after every read?  If so, you should not, only call .Close() when you are done reading/writing.
What PLC are you working with?

velmurugan1, 2018-03-12:
Dear sir

 Here attached the error page

kindly give any solution.

Thank you..........

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

*M.VELMURUGAN*

*22(2), GK Square ,1st Street , Kurchi,Sundarapuram, Sundarapuram, Kurichi,
Coimbatore, Tamil Nadu 641024*
*Ph:9942133658*

dmroeder, 2018-03-12:
I am unable to view your error, it appears that Avast has had it's way with it.

Please copy and paste the message directly here.  Please also address the other questions I had.

velmurugan1, 2018-03-12:
Dear Sir

Tracebac{ (most recent call last)
file"F:/mycer/ANN/pylogix-master/multionetime.py",line 11,in <module>
ret comm.MultiRead("A","B","C","D")
file"F:/mycer/ANN/pylogix-master/multionetime.py",line 110,in MultiRead
return _multiRead(self,args)
file"F:/mycer/ANN/pylogix-master/multionetime.py",line 261,in _multiRead
IntialRead (self, t,b)
file"F:/mycer/ANN/pylogix-master/multionetime.py",line 1153,in IntialRead
timeout:timed out

   - *PLC=ABB(compact logix compactlogix l33er)*

from eip import PLC
import time

#def ex_read(tag):
a=0

while True:
    comm = PLC()
    comm.IPAddress = '192.168.1.1'

    ret = comm.MultiRead("A","B","C","D")

    print  ret

    #time.sleep(1.0)

    a=a+1
    print a

this is programe

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

*M.VELMURUGAN*

*22(2), GK Square ,1st Street , Kurchi,Sundarapuram, Sundarapuram, Kurichi,
Coimbatore, Tamil Nadu 641024*
*Ph:9942133658*

dmroeder, 2018-03-12:
One problem I see is that you are creating a new instance of PLC with each read.  Try this instead:

from eip import PLC
import time

with PLC() as comm:
    comm.IPAddress = '192.168.1.1'
    
    while True:
        ret = comm.MultiRead("A","B","C","D")
        print ret #print all the returned values
        time.sleep(0.5)

dmroeder, 2018-03-12:
for some reason, I can't get the formatting right tonight...

velmurugan1, 2018-03-12:
Dear Sir

Are you saying about time.sleep or with PLC() as comm:,becuse  i have
already checked with time.sleep 0.5 and time.sleep 1.0.

When i put 0.5 it goes 5 min without error

When i put 1.0 it goes to 10 min with out error

Thank you

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

*M.VELMURUGAN*

*22(2), GK Square ,1st Street , Kurchi,Sundarapuram, Sundarapuram, Kurichi,
Coimbatore, Tamil Nadu 641024*
*Ph:9942133658*

dmroeder, 2018-03-12:
I'm saying with your original code, you are creating an instance of PLC() each iteration of your while loop.  This is not the way to do it.  You need to create an instance of PLC(), set it's IP address, then in your while loop, read your tags.

Did you try the example I provided above?

dmroeder, 2018-03-12:
Could I convince you to capture the error using Wireshark and email it to me?

velmurugan1, 2018-03-13:
Dear sir

Here this PLC now not available i will contact after few days

Thank you

dmroeder, 2018-03-15:
Could you try a test for me when you get some time?  Search for socket.settimeout and change both of them to socket.settimeout(5).  This will allow for 5 seconds to receive data.  See if that makes it more stable.  If you do happen to wireshark the problem, please send it my way.

closed 2018-06-04 by dmroeder

velmurugan1, 2018-06-05:
Dear Sir

We have expected get the data from Structure from compact logix kindly
provide sample.

Thank you

*M.VELMURUGAN*

*22(2), GK Square ,1st Street , Kurchi,Sundarapuram, Sundarapuram, Kurichi,
Coimbatore, Tamil Nadu 641024*
*Ph:9942133658*

dmroeder, 2018-06-05:
There is no solution.  You must read each element of your structure individually.

velmurugan1, 2018-06-09:
Ok thank you very much

*M.VELMURUGAN*

*22(2), GK Square ,1st Street , Kurchi,Sundarapuram, Sundarapuram, Kurichi,
Coimbatore, Tamil Nadu 641024*
*Ph:9942133658*