dmroeder/pylogix #14: Identifying which tags are multidimensional?

jdingus, 2018-04-11:
Just want to say thank you for this library! 
I have a project which I will be going up to 6 or 7 plcs and doing some data monitoring.
Using your library is there a method for determining which tags have an array rather than a single variable?  Working on a script which provide a live update of the variables as the machine is running.  Thanks again for all your work on this library.

dmroeder, 2018-04-12:
There is not a way.  I'm curious though...

Can you give me a bit of context on what you are planning on doing? Maybe with tag name examples and datatypes?

Are these tag names that you already know, but the same tag is an array in one PLC but not in another?
Or are you using GetTagList to get tag names, then trying to read them?

dmroeder, 2018-04-12:
Ah I see.  This will be much more difficult.  It is not impossible, but not currently possible.  On my list of things to work on, I have always thought it would be cool to do exactly what you are after, just connect to the PLC, get the tags, and start reading them.

You are correct, the PLC has arrays, and multi-dimensional arrays (up to 3 dimensions).  Using more than one dimension is a bit uncommon though, but I suppose that doesn’t matter.

A Logix PLC has a few basic data types:  STRING, SINT, INT, DINT, LINT, REAL, BOOL.  Then there is the UDT.  UDT is a user created data type, or a data type that is made up of the base data types.  When requesting the tags, I get the tag name and basic data type.  The problem that I need to solve is interrogating the UDT’s to get down to their basic data type.

Let’s say we have a UDT called “Person”, it’s made up of a “Name”(STRING) and an “Age” (SINT).  So we create an instance of it and we have Bill.Name and Bill.Age.  When you request the tag list, you will get Bill, and it will be identified as a UDT.  A tag called MyTimer that is the TIMER data type also appears to be a UDT.  What would need to happen is we would need to find out the makeup of “Person”.  Then you add the possibility of nested UDT’s and arrays, it gets more complicated.  And the process takes a bit longer.

This is something I hope to work out as time allows.  One user shared a clever way of working with UDT’s which involved exporting the PLC program to a L5X file (which is essentially XML).  You can fairly easily parse a L5X in python to get the tag names and structures, then read the tags that were in the L5X. This may be an option for you for now.  I'll think some more about this...

jdingus, 2018-04-15:
Thanks for info on parsing the L5X export.

Would work just fine and already some good libraries developed.

dmroeder, 2018-04-15:
Hopefully I'll sort out getting a more complete picture of the tags soon. I've been experimenting with some ideas.  Stay tuned...

su600, 2019-09-26:
Thank you for this library too! 
Is there any progress on this issue? Can we indentify the dimensional of Tags now ?

kodaman2, 2020-03-11:
Could this be done by brute force?

try a tag[0], if it returns something, then try tag[1], and just keep going until you get None. That's one way to find out the depth of a tag.

Thoughts?

ottowayi, 2020-03-12:
You can get the dimensions of a tag by reading attribute 8 for the Symbol Object.  [Here](https://github.com/ottowayi/pycomm3/blob/master/pycomm3/clx.py#L231) is how I do it.

dmroeder, 2020-03-21:
Thanks for the details @ottowayi, I'll check into it.

th3dj, 2020-03-31:
I just used the l5x parsed it and read the Demention tag and then a 
`ref = comm.Read('tag', dementionvariable)`

That way it just pulls the whole array. My first try used a while loop but it wasn't to my liking and lists are easier to work with plus i don't have to assemble all the output form the read.