dmroeder/pylogix #17: Library Download and usage

chrisegge, 2018-05-09:
I know this isn't an issue with the library you have provided. I am completely new to python and I am trying to get this going on windows 7. I have downloaded python v 2.7 and started doing a lot of research and practice on the coding part but I have no idea how to get this library into python so I can begin writing code. To use this code to get the tag information. Any information would be greatly appreciated because I am struggling greatly. I am a skilled plc programmer but it is a big jump from ladder logic to structured text. Thank you in advance.

dmroeder, 2018-05-09:
Hey chrisegge, shoot me an email, I'd be glad to help.

closed 2018-05-10 by dmroeder

leandroha, 2019-04-29:
I'm having the same problems.

dmroeder, 2019-04-29:
Feel free to email me. dmroeder@gmail.com.

evaldes2015, 2019-04-29:
Chrisegge,

Why are you starting with Python 2.7? Is there a specific reason? I think you would be better off with 3.7.

dmroeder, 2019-04-29:
evaldes2015, Chris opened this issue last year, leandroha commented earlier today.  Necropost.

evaldes2015, 2019-04-29:
Opps, my bad. Didn't look at the dates. Sorry.

leandroha, 2019-04-29:
Hi, I didn't know Git, but now I got it.
I've installed pylogix and i'm trying use it. I just don't understood the
dataype. You use a [tagName, numberDataType]. I don't know what is that
number. Datatypes are like "INT", "DINT", "D2SD".
If i want to read all tags from D2SD data type, how can I do it?

à(s) 12:18:

dmroeder, 2019-04-29:
I assume D2SD is a UDT.  Reading UDT's is not exactly supported.  You will need read each member of your UDT individually.