dmroeder/pylogix #18: MultiRead timing out

mainsteam, 2018-06-01:
Hi!
I'm able to read individual tags with no issue, but when I try to do a MultiRead, I get the following error:
```python
>>> tags1
['HEDS.Shift_Number', 'HEDS.Status', 'HEDS.Shift_Hours']
>>> values1 = t1.MultiRead(*tags1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 109, in MultiRead
    return _multiRead(self, args)
  File "eip.py", line 260, in _multiRead
    InitialRead(self, t, b)
  File "eip.py", line 1152, in InitialRead
    retData = self.Socket.recv(1024)
socket.timeout: timed out
```

Additionally, when I was trying to generate some sample output for this post, then second time I tried within the same python session, this was the result:
```python
>>> tags1[0]
'HEDS.Shift_Number'
>>> v1 = t1.Read(tags1[0])
>>> print(v1)
2
>>> tags1
['HEDS.Shift_Number', 'HEDS.Status', 'HEDS.Shift_Hours']
>>> values = t1.MultiRead(*tags1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 109, in MultiRead
    return _multiRead(self, args)
  File "eip.py", line 260, in _multiRead
    InitialRead(self, t, b)
  File "eip.py", line 1162, in InitialRead
    raise Exception('Failed to read initial tag: ' + cipErrorCodes[status]) 
KeyError: -1
```

Any thoughts? I've already increased the timeout value (socket.settimeout) to 10 (line 484)

Thanks for any help you can provide!

dmroeder, 2018-06-01:
Interesting.  So you can read each of those tags individually, but it fails using MultiRead?
What PLC model are you using?
What data types are those tags?
Can you email me a wireshark trace when this happens?

mainsteam, 2018-06-01:
So I couldn't get wireshark running in my linux VM quickly, but I'll get back to that if we find it ends up being necessary. For your other questions:
PLC: CompactLogix
Datatypes: Int, Int, Float

I did some additional troubleshooting, trying some other things I thought might be diagnostic and here's what I found:

```python
Python 2.7.15 (default, May  1 2018, 20:16:04) 
[GCC 7.3.1 20180406] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import eip
>>> tags = ['HEDS.Shift_Number', 'HEDS.Status', 'HEDS.Shift_Hours'] 
>>> test1 = eip.PLC()
>>> test1.IPAddress = '10.24.5.68'
>>> for i in range(len(tags)):
...     value = test1.Read(tags[i])
...     print(value)
... 
2
1
2.46288371086
>>> values = test1.MultiRead(*tags)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 109, in MultiRead
    return _multiRead(self, args)
  File "eip.py", line 285, in _multiRead
    status = unpack_from('<h', retData, 48)[0]
struct.error: unpack_from requires a buffer argument
>>> values = test1.MultiRead('HEDS.Shift_Number', 'HEDS.Status', 'HEDS.Shift_Hours')
>>> values
[2, 1, 2.5986595153808594]
>>> values = test1.MultiRead('HEDS.Shift_Number', 'HEDS.Status', 'HEDS.Shift_Hours')
>>> values
[2, 1, 2.6017134189605713]
>>> values = test1.MultiRead(*tags)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 109, in MultiRead
    return _multiRead(self, args)
  File "eip.py", line 285, in _multiRead
    status = unpack_from('<h', retData, 48)[0]
struct.error: unpack_from requires a buffer argument

```

So to me it looks like MultiRead is actually working, but it can't handle list unpacking for some reason. I can work around this if I have to, but it's strange to me because I thought that with the way list unpacking worked, the recieving function shouldn't be able to tell the difference between a directly entered list and an unpacked one.

Interested to hear what you think of this.
Thanks!

dmroeder, 2018-06-03:
The error you are getting doesn't have to do with you passing the list of tags to MultiRead(), it has to do with the amount of data that is coming back in the packet.  It's trying to unpack the status of the reply at byte 48, but the packet isn't that long for some reason.

I think wireshark would be helpful if it would be possible for you to capture it.  I cannot recreate your problem.  Passing the list of tags the way you are works just fine.  I wrote a script that mimicked yours, multiread every 100ms and ran it for an hour.

If you can get a wireshark of it, you can email it to me and I'll take a look.

dmroeder, 2018-06-04:
Disregard,  you don't need to send me a wireshark trace.  If you get time, try the latest commits.  I was able to recreate what I think was a similar problem.

mainsteam, 2018-06-05:
So far all the tests I've done are working just fine. I think you've solved it! Thanks for your help.

closed 2018-06-05 by mainsteam