dmroeder/pylogix #19: Inconsistent MultiRead errors

mainsteam, 2018-07-07:
Hello,
I'm seeing some errors while using MultiRead. I do not have issues if I loop single reads over the list of tag names. For example:

```python
>>> test1 = eip.PLC()
>>> test1.IPAddress = cell1.ipAddr
>>> data1 = test1.MultiRead(*tags)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 109, in MultiRead
    return _multiRead(self, args)
  File "eip.py", line 285, in _multiRead
    status = unpack_from('<h', retData, 48)[0]
struct.error: unpack_from requires a buffer of at least 2 bytes
>>> data1 = []
>>> for i in range(len(tags)):
...     data1.append(test1.Read(tags[i]))
... 
>>> data1
[2, 2, 6.210262298583984, 1.1434727907180786, 0.19440944492816925, 4.872259616851807, 85, 0.0, 13.68673324584961, 0.0, 0.0, 157404, 48.4294319152832, 0.0, 0.0, 0.21454083919525146, True]
>>> data1 = test1.MultiRead(*tags)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 109, in MultiRead
    return _multiRead(self, args)
  File "eip.py", line 285, in _multiRead
    status = unpack_from('<h', retData, 48)[0]
struct.error: unpack_from requires a buffer of at least 2 bytes
```

For multiple PLCs I'm able to use MultiRead with no issue, but they are all CompactLogix with Ethernet built into the processor. The two PLCs I'm having issues with are different, however. They are a 1768-L43 and a 1756-L75 with separate ethernet cards.

dmroeder, 2018-07-07:
Can you email me the exact script you are using and possibly a wireshark of the issue?
Can you give me some context?  Like number of tags you are reading, data types, etc.  Does this happen every read, or every 10th read or every 100th read, etc.  Maybe mention the firmware of your L75 and the model and firmware of the Ethernet module in your CLX chassis.

I don't have a 1768 series but I do have a L73 at work, I'll have access to it Monday.

mainsteam, 2018-07-07:
Sure, what address should I send it to? Thanks! I'll check on the reduced tags.

dmroeder, 2018-07-07:
dmroeder@gmail.com.  Thanks

mainsteam, 2018-07-08:
I've sent you a .pcap file. I have not yet had a chance to try again with fewer tags, but I'll let you know when I do. The pcap file is from running the example I posted above in the interactive console. It happens every MultiRead. I have a workaround in my program running right now which is that for those two PLCs, I loop through the tag list with single Reads and it's working fine for now. The data types are ints, floats, and a bool. Thanks for your help!

mainsteam, 2018-07-13:
Looks like your analysis from our email conversation was correct. So far, I've split all my MultiReads into two sets and the errors haven't recurred.
For others who might be reading, there is a 500 byte limit on tag names in a request, so if you have too many long tag names in one MultiRead, it'll error out. Thanks!

kyle-github, 2018-07-13:
IIRC, if you either use unconnected messaging or your PLC only supports ForwardOpen, but not ForwarOpenEx, you only get a payload size of ~500 bytes.   It might be a few more, but you can only express 511 in the ForwardOpen request.

dmroeder, 2018-07-13:
Thanks for the update mainsteam.  

Just to add a little info on why this appeared to only be an issue on CLX and the L43.  Although the UDT was the same structure in all the PLC's, the base tag name was longer in the CLX and L43, which made the packet longer.  I'll look at a way to handle this more gracefully. 

Kyle, thanks for the input.  I'll do some experimenting with unconnected sends.

closed 2018-09-28 by dmroeder