dmroeder/pylogix #20: Using this library with Omron PLC

Dhiaaeddinanabtawi, 2018-08-01:
Hi,

I am trying to use this library with an Omron PLC. I want to get the PLC time just to make sure the communication works. I am getting the following error:
`File "C:\Python26\pylogix-master\plcread.py", line 5, in <module>
    comm.GetPLCTime()
  File "C:\Python26\pylogix-master\eip.py", line 115, in GetPLCTime
    return _getPLCTime(self)
  File "C:\Python26\pylogix-master\eip.py", line 336, in _getPLCTime
    raise Exception('Failed to get PLC time, ' + err)
Exception: Failed to get PLC time, Path destination unknown
`
It seems like the PLC gets the request and replies, but it is not sending back any values. My thinking goes to the initial parameters in `init` function. I changed `VendorID` to match Omrons ID and I changed `SerialNumber` to Omron's serial number. What else I have to change to make the PLC give me back its time?
Maybe I am mistaken and it is completely something else that causes the problem? 

Thanks in advance
Dhiaa Eddin Anabtawi

dmroeder, 2018-08-01:
First of all, this library only works with Rockwell Automation PLC's, specifically CompactLogix, ControlLogix and Micro820/850 PLC's.   It will not work with Omron.

Second, the VendorID is not really relevant this type of connection.

Third, the serial number is a randomly generated value used to identify the connection, it is not the serial number of a device.

Last, while Omron may make Ethernet I/P adapters/scanners, they are most likely intended for an I/O type connection, which this library does not do.

closed 2018-08-01 by dmroeder

Dhiaaeddinanabtawi, 2018-08-14:
Is there any way to make it work with Omron PLC's as well?

dmroeder, 2018-08-14:
Not by me unfortunately.  I have no experience with Omron and I don't have access to any of their hardware.   I recommend you check out AdvancedHMI, it has a native communication drivers for Omron.

Dhiaaeddinanabtawi, 2018-08-14:
Ok, Thank you for your answer and help. Have a nice day!

dmroeder, 2018-08-14:
You do the same!  Hopefully you'll find AdvancedHMI useful.  They have a good forum too if you need pointers.  I post over there occasionally.