dmroeder/pylogix #21: Determining if connection is still lost

mhuckaby21, 2018-08-19:
Once the socket is connected, how can you determine if the PLC goes away?

I get an unpack_from requires a buffer argument error from the Read function when I attempt to read after the PLC goes away (line 165 in the eip.py file).  The reData value would be empty and the unpack_from function requires data to be present.

I'm only reading values every few minutes.  I don't know if it would be better to ping the PLC IP, restart a connection, build a function to see if the PLC is still present, or build a better error trap in the Read function.

Thanks!

dmroeder, 2018-08-19:
There is definitely room for improvement on error handling on my part.  For now, if you are only reading every few minutes, I'd do your reads, then close the connection by calling .Close().  Even though you are probably not currently closing the connection, the PLC will purge connections after a period of inactivity.

I'll see what improvements I can make.

mhuckaby21, 2018-08-20:
No problem.  I just wanted to make sure that I wasn't overlooking something.

If I come up with a solution I'll let you know.

Thanks!

tblount70, 2020-02-28:
Has anyone made progress on this issue? My python is pretty limited, but I'm thinking of jumping into this. My application is a 4-line text display replacement. So it's pretty simple. The issue is that the my users need some indication that the system is no longer communicating.

dmroeder, 2020-02-28:
I would say that it has been improved since the addition of returning the Response class.  If you are actively reading something in a loop, then suddenly the PLC is no longer available, you will initially receive "Connection Lost" as the status.  The problem then is that the next read, it will try to reconnect and will fail.  If you use a try/catch around your read, it will reconnect once the PLC is available.

tblount70, 2020-03-01:
Excellent. I just added try/except and it works perfectly. If I were a little more experienced with python I guess I would have known to try that.

For my simple application this is probably all I'll ever need in the way of comm loss detection.

Thank you!

marne44, 2020-03-02:
> Excellent. I just added try/except and it works perfectly. If I were a little more experienced with python I guess I would have known to try that.
> 
> For my simple application this is probably all I'll ever need in the way of comm loss detection.
> 
> Thank you!

Can you please upload and commit?

Thanks in advance.

kodaman2, 2020-03-02:
@marne44

Try/Catch is a concept in all programming languages. Have a look in my repo file below line 158 fir an example of try catch. Basically what happens is that it tries to do the code inside the try, if it gets an exception then the code inside the exception gets executed. There's another one finally, for clean up too. 

https://github.com/kodaman2/Data_Preserve/blob/master/data_preserve.py

tblount70, 2020-03-02:
This is what's working for me...

def UpdateValue():
    try:
        retL = comm.Read(tagList1)
        myapp.commerror=FALSE
    except:
        myapp.commerror=TRUE
            
    ### ... continue processing valid results

closed 2020-03-11 by kodaman2