dmroeder/pylogix #22: Reading Tags From a SLC 500 or Micrologix

GearDownForWhat, 2018-08-27:
I realize that this library is not intended for reading from these processors, but i was wondering whats really preventing it? 
the communications seems like it should be similar, I.E. instead of reading a tag named "MyTag4" you would read/write a tag named "N7:0/0"

I actually gave it a try, but it was a no-go.

dmroeder, 2018-08-28:
It may seem like it should work that way, but not quite.  A lot of what would be necessary is there, it would take some new code to properly parse and pack the request into the packet, also parse and unpack the reply.   Very doable, but I don't have access to very much "logix 500" based hardware, or a lot of extra time to test and maintain it.

GearDownForWhat, 2018-08-28:
understandable, thanks for taking your time to answer, and thank you so much for creating this amazing bit of software. 
I was EXTREMELY surprised when I grabbed it and tried it and was reading and writing tags in less than a half hour with almost no python experience, so thank you for making this so easy!