dmroeder/pylogix #23: Communication with Emulate 5000

raincls, 2018-09-01:
When there is no plc hardware, I try to simulate the PLC with Emulate 5000.
But the program seems to be unable to discover any Ethernet I/P devices.
May I ask can Emulate 5000 simulate the communication process?
Thanks for your time.

dmroeder, 2018-09-01:
This will not work with Emulate since Emulate does not support Ethernet I/P communications.  However SoftLogix does if you have access to it.

When you say that you are unable to discover devices, are you basing that on trying to discover Emulate?  As I mentioned, it does not support Ethernet I/P.  I have no problems discovering devices.

raincls, 2018-09-02:
I was indeed trying to discover Emulate.
Thank you very much for your careful explanation. I will try SoftLogix.

closed 2018-09-02 by dmroeder

dmroeder, 2018-09-02:
Sure no problem.  Keep in mind, with SoftLogix, you will have to specify the processor slot since RSLinx goes in slot 0 (for some dumb reason).

kodaman2, 2018-11-02:
Were you able to get it working with SoftLogix? I am able to discover the SoftLogix Ethernet module but returns 0.0.0.0 for the IP even though is set for 192.168.0.27
Anything special I should be looking for? I have slot 2 for the cpu.

kodaman2, 2018-11-03:
Strange, it works now. Don't know what exactly fixed it. I read on SoftLogix manual about disabling UDP packets in Rslinx, so I did that. Reboot and ran discover function, and it found SoftLogix this time with the correct IP. I ran getTagList and also pulled all tags. I went back and enable UDP again, reboot and tested again with no issues. Looks like I am good to go. Yeah definitely something got hung, either a service or a process.

dmroeder, 2018-11-03:
Classic Rockwell.  Thanks for the update.

dmroeder, 2021-08-19:
While I know this is old, and not likely to reach anyone, I cannot let misinformation go.  I found out recently that you can use pylogix with Emulate.  A user confirmed that with 0.7.14, set ConnectionSize to 504 and the IP address will be your computer.

TheFern2, 2021-08-20:
@dmroeder I couldn't get a read with Studio 5000 Emulate V32, maybe it was successful with the old RS Emulate. 
Do you have a version where it was known to work? I tried all local IPs, and also different slots, on the Emulator the CPU appears to be on slot 2.

```
= RESTART: C:\Users\Default.DESKTOP-GU2MB4F\Downloads\pylogix-master (2)\examples\01_read_simple.py
Traceback (most recent call last):
  File "C:\Users\Default.DESKTOP-GU2MB4F\Downloads\pylogix-master (2)\examples\01_read_simple.py", line 21, in <module>
    ret = comm.Read('BaseBOOL')
  File "C:\Users\Default.DESKTOP-GU2MB4F\AppData\Roaming\Python\Python39\site-packages\pylogix-0.7.15-py3.9.egg\pylogix\eip.py", line 107, in Read
    return self._readTag(tag, count, datatype)
  File "C:\Users\Default.DESKTOP-GU2MB4F\AppData\Roaming\Python\Python39\site-packages\pylogix-0.7.15-py3.9.egg\pylogix\eip.py", line 252, in _readTag
    resp = self._initial_read(tag, base_tag, data_type)
  File "C:\Users\Default.DESKTOP-GU2MB4F\AppData\Roaming\Python\Python39\site-packages\pylogix-0.7.15-py3.9.egg\pylogix\eip.py", line 1366, in _initial_read
    status, ret_data = self.conn.send(request)
  File "C:\Users\Default.DESKTOP-GU2MB4F\AppData\Roaming\Python\Python39\site-packages\pylogix-0.7.15-py3.9.egg\pylogix\lgx_comm.py", line 65, in send
    return self._getBytes(eip_header, connected)
  File "C:\Users\Default.DESKTOP-GU2MB4F\AppData\Roaming\Python\Python39\site-packages\pylogix-0.7.15-py3.9.egg\pylogix\lgx_comm.py", line 150, in _getBytes
    raise BufferError("ethernet/ip _getBytes output size exceeded: %d bytes" % len(data))
BufferError: ethernet/ip _getBytes output size exceeded: 64 bytes
```

dmroeder, 2021-08-20:
Hmm, I believe the version was 0.7.14, all they did was set ConnectionSize=504.  Apparently using loopback address worked and "the address that shows up in RSLinx", which I assume was the computers address.  The user called the software package "Emulate 5000" which leaves an opening to be ether the old or new.

I don't think that software is part of our toolkit, so I can't really test myself.  Can you wireshark it?

TheFern2, 2021-08-20:
It works after locally changing connection to 504, I found the bug the ConnectionSize is now set to 50 in lgx_comm.py must have accidentally deleted the 4. Both loopback and computer ip worked. Nice!

[image]

[image]

dmroeder, 2021-08-20:
I am an idiot.  looks like my 2nd commit, I accidentally changed the fall back connection size.  I gotta be more careful than that.  I was in a hurry, I usually check diff before committing.

TheFern2, 2021-08-20:
> I am an idiot. looks like my 2nd commit, I accidentally changed the fall back connection size. I gotta be more careful than that. I was in a hurry, I usually check diff before committing.

no worries I am gettng fixed, and adding some notes to the docs as well. It only appears to work locally.

dmroeder, 2021-08-20:
> > I am an idiot. looks like my 2nd commit, I accidentally changed the fall back connection size. I gotta be more careful than that. I was in a hurry, I usually check diff before committing.
> 
> no worries I am gettng fixed, and adding some notes to the docs as well. It only appears to work locally.

Yeah I had heard that it only works on the machine that is actually running emulate.

sagar-gulve, 2023-04-28:
While trying this I am getting strange error _**Unknown error [WinError 10061] No connection could be made because the target machine actively refused it**_

Any suggestion on this?

sagar-gulve, 2023-04-28:
@dmroeder : Steps mentioned on issue#221 resolved it. Thanks..!!