dmroeder/pylogix #24: Accessing Task Specific Tags

nrjbs87, 2018-09-16:
I am able to read/write to controller tags in the main task no problem. Is there a syntax to read/write to tags in subtasks?

dmroeder, 2018-09-16:
What you are after is program scoped tags, I thought I had that in the README, but it appears that I do not.  It's like this:

Program:ProgramName.TagName

So let's say your program name is "Line1" and your tag name is "ProductCount".  It would look like this:

Read("Program:Line1.ProductCount")

nrjbs87, 2018-09-16:
Definitely a step in a the right direction as this is what I was looking for. I am fairly new to PLCs so I need a bit more help. 

Currently I have my main Program which we'll call Program. After this, I have numerous Tasks each with different phase types. And then within the phase types, I have several instances. And each of those phase instances I have scope tags I am looking to reference. 

So really I am structured as such:

Program
Tasks
      Phase Type
              Phase
                   Tag I need

Is it possible to reference the Tag I need with this structure?   

Thank you.

dmroeder, 2018-09-16:
I see, so I guess I first should have asked if the tag you are after is a local tag to the Equipment Phase, or it is one of the automatically generated phase tags.

For tags local to the Equipment Phase, these are treated the same as Program Scoped tags, so the syntax that I illustrated before will work.

For tags that are generated when crating an Equipment Phase, those appear to exist in the controller scope.  So let's say I created a EP called "Line1".  If I look in the controller scoped tag database, I will see a tag called Line1 and it will have the data type PHASE.  I can the "Running" tag for line 1 by reading "Line1.Running".  Or state by reading "Line1.State".  

One thing that may help you is by going into the controller tags, looking right above the tag window for the "Scope" drop down.  This will let you view all of the tags in your program according to the scope that they are created in.  This may help you with where they exist and how you need to access them.  Any tag that is not located in the controller scope will need to be accessed with the format I used in my first reply.  

Hopefully that makes sense, if not, let me know where I can clarify.

closed 2018-12-10 by dmroeder