dmroeder/pylogix #25: Controller firmware version data

kodaman2, 2018-09-21:
It is possible to get controller firmware version? or controller model?

dmroeder, 2018-09-22:
The only way to do that at the moment is to use .Discover(), then iterate through the devices in the reply, you should be able to get the information you need based on the IP Address.

I suppose the flaw with that though would be with CLX systems, since it would discover the Ethernet Module and not the controller itself.  I'll get something going to get the properties of a specific device or module.

kodaman2, 2018-09-22:
I'll give it a try and see what information I get with Discover() method.

dmroeder, 2018-12-14:
I added GetModuleProperties() to get the properties of a module.  Just specify the slot number of the properties you want.  It returns the same type as Discover().   So in the reply, .ProductName and .Revision for the example you asked about.  Let me know what you think.

kodaman2, 2018-12-19:
Dustin this worked great, ran the GetModuleProperties on 6 slots. Thanks for the addition! This issue/enhancement can be closed for sure : )

```
1756-L84E/B  -  Rockwell Automation/Allen-Bradley  -  30.11

1756-L73/B LOGIX5573  -  Rockwell Automation/Allen-Bradley  -  30.11

1756-EN2T/C  -  Rockwell Automation/Allen-Bradley  -  4.3

1756-EN2TR/C 217021900  -  Rockwell Automation/Allen-Bradley  -  10.7

1756-EN2TR/C 217021900  -  Rockwell Automation/Allen-Bradley  -  10.7

MVI56E-MNETC  -  Prosoft Technology  -  3.1

MVI56E-MNETC  -  Prosoft Technology  -  3.1
```

dmroeder, 2018-12-19:
Nice, thanks for the update.  That is a high dollar setup you have there!

closed 2018-12-19 by dmroeder