dmroeder/pylogix #27: Failed to register session

riazmoradian, 2018-10-26:
Hi,

While trying to connect using your instructions, I get the error "Failed to register session" for both lines labeled 1 and 2. Could you help me?

I guess it might be an authorization issue, since the PLC Server has a username and password set. Are there any authorization methods in this module? If so, I was unable to find them.

Code - 

```
from eip import PLC
test = PLC()
test.IPAddress = "192.168.3.3"
test.Port = 4048
print test.GetTagList()  #1
print test.GetPLCTime()  #2
```

dmroeder, 2018-10-26:
I'm not sure I follow your "authorization" problem but you shouldn't be changing the port, leave it at it's default 44818.

What PLC model are you trying to connect to?

dmroeder, 2018-10-31:
I hate to leave this unresolved, any thoughts on my previous comment?

Can you confirm that you were trying to read data from a CompactLogix, ControlLogix or MicroLogix 8xx PLC?

riazmoradian, 2018-11-14:
Hey, I'm sorry I took so long to reply.

Our focus shifted and we aren't planning to read data from a PLC anymore.
The error was because the IP address wasn't a PLC, it was an OPC server, I
had the wrong info.

You can go ahead and delete the thread if you want. Sorry once again!