dmroeder/pylogix #29: test.Read("tagname") returns None

usaish, 2018-10-29:
HI,

First, thank you for this amazing library.

As mentioned, I am able to connect to the PLC and extract the tag list. However, when I try to print a tag value, it shows None. e.g.

>>> value = test.Read("testbit")
>>> print (value)
None
>>> 
[image]

dmroeder, 2018-10-29:
This is a sign that it is not actually connecting to the PLC.  If you would perform your test while running Wireshark, I can look over the file and verify, you can email it directly to me.

It should be telling you the problem though, I'll make some changes here to better inform users.

dmroeder, 2018-10-31:
I pushed a change that should raise an exception if the initial connection to the PLC fails, rather than return None.  

Typically what you were seeing was a result of the PLC IP address being wrong, or the network adapter not being available (disconnected or disabled).

usaish, 2018-10-31:
You are right. The plc wasn’t communicated. I restarted the comms and it worked. Thanks for adding the exception

dmroeder, 2018-10-31:
Great thanks for the update!

closed 2018-10-31 by dmroeder