dmroeder/pylogix #32: Connection Lost

naterbug25, 2018-11-11:
Hello,

I am using a Raspberry Pi 3B+ & Compact Logix L23E PLC and I am having issues connecting the two.

The variable "x1" is a bool tag that has External read/write access.
I can ping the PLC from the Pi.
I have not added any Ethernet modules to the PLC for communication to the Pi. 

Any suggestions would be great. Thanks!

Ran on the Raspberry PI:

pi@raspberrypi:~/pylogix $ python
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from eip import PLC
>>> test = PLC()
>>> test.IPAddress = "192.168.1.155"
>>> value = test.Read("x1")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "eip.py", line 90, in Read
    return _readTag(self, args[0], 1)
  File "eip.py", line 160, in _readTag
    InitialRead(self, t, b)
  File "eip.py", line 1175, in InitialRead
    raise ValueError(cipErrorCodes[status])
ValueError: Connection lost
>>>

dmroeder, 2018-11-12:
Are you absolutely certain that 192.168.1.155 is your PLC's IP address?

Since you get "Connection lost" on the initial read, it suggests that a connection was made.  The only way I have ever been able to reproduce this is if I try to connect to a device that understands Ethernet I/P but is not a PLC (Point I/O comm module for example).

If you could capture this with Wireshark and email it to me, that might be helpful.

BTW, nothing has to be added to the tree for pylogix

naterbug25, 2018-11-13:
I am sorry I raised this as an issue... I spent more time than I want to admit troubleshooting this. I now have it working. I had the wrong IP Address. 

Thank you for your help!

dmroeder, 2018-11-13:
No worries, glad it's working, thanks for the update.

closed 2018-11-13 by dmroeder