dmroeder/pylogix #33: Unexpected Individual Read Result from Array

kodaman2, 2018-12-20:
Question: I have a file with 18000 or so tags. I am reading the status and then writing that to a file one by one.  I've found something strange the first 300 tags are as follows:

```
Program:MyProgram.MyTag[0]
Program:MyProgram.MyTag[1]
...
Program:MyProgram.MyTag[299]
```

I am doing the read tags as follows:
`test.Read(tag)`

If I run my program it will raise an exception "Read failed Unkown Error" exactly at:
`Program:MyProgram.MyTag[160]`

My first thought was that maybe I was reading the tags too quickly, so I added a sleep(1) in between reads but exactly at 160 the exception is thrown.

I figure maybe there's a limit, tags from 0 to 159 are read fine is  boolean and I print out to the console to check True and False were coming across. As a test I deleted all tags from this array starting at [160]. I checked the whole file and no other array went over 160. After deleting 160-299 lines from that array tag, I ran the program and it executed fine read all tags and wrote to the file.

Am I overlooking something? Have you come across this before?

dmroeder, 2018-12-20:
No you are not overlooking and no I have not come across this before.  I see what is going on though, I have an issue with how I'm handling my nemesis, BOOL arrays.  I'll get it fixed and give you some details on what was going wrong.

dmroeder, 2018-12-21:
Try the latest commit, let me know what you think

kodaman2, 2018-12-21:
Nice, issue fixed! Complete boolean array of 300 indices was read with no problems.

closed 2018-12-21 by kodaman2