dmroeder/pylogix #35: Regards Working for All plc and working for raspberry pi 3

SalmanSubetha, 2018-12-22:
hello sir i have worked in vipa 015 plc is it suitable for this code and how to configure the files in raspberry pi and how to configure to vipa controller to raspberry pi 3 using ethernet ip

kodaman2, 2018-12-23:
This library is for Allen Bradley PLCs. If I understand correctly VIPA uses step7 protocols. If so you should be looking at Libnodave library. There is an implementation in python but I haven't used it before in python, only in C#. Please look below for details on snap7, good luck. 
[http://snap7.sourceforge.net/](snap7)
https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PLC/VIPA.htm

dmroeder, 2018-12-26:
As kodaman2 mentioned, this is for CompactLogix, ControlLogix and Micro8xx PLC's only.

closed 2018-12-26 by dmroeder