dmroeder/pylogix #37: Multi-read failed

Lucasrsv1, 2019-02-06:
Hi, thank you for this awsome library, it is helping me a lot!

But I've been having an issue with multi-reads. I'm using the PLCs Micro850 and CompactLogix L18ERM, but multi-read doesn't seem to work in either of them. On CompactLogix I get the message "Multi-read failed, Embedded service error" and on Micro850 I get "Multi-read failed, Service not supported". Currently I'm calling the function Read with only an array of tags as parameter.
When I read each tag individually it works fine, but I need to read lots of data all the time, so multi-read would be really important.

Do you have any clue about why multi-read is not working for me?

Thanks in advance.

dmroeder, 2019-02-06:
So multi-read doesn't work with the Micro8xx because they don't support that feature.  Keep in mind,  there is a limit to how many tags you can read at once.  This limit is packet size so the data type matters.  I usually suggest reading 10 tags at a time.

Since we rule out the Micro8xx, let's discuss the L18.  Can you share some details about the tags you are reading, how many, types, etc?  Maybe a snippet of your code?

Also, I was going to remove .MultiRead() at some point since .Read() will now do the same thing if you pass it a list.

Lucasrsv1, 2019-02-06:
Thank you for the fast reply!
It's good to be sure that Micro8xx really doesn't suport multi-read.
So, about the ControlLogix, I was trying to read 4 tags at the same time ["bool01", "int01", "float01", "string01"], and the string has 82 characters. My connection code looks like this:

`from pylogix.eip import PLC
class PLCMonitor:
	def __init__(self):
		self.pylogixPLC = PLC()
		self.pylogixPLC.IPAddress = "192.168.1.1"
		self.pylogixPLC.ProcessorSlot = 0
		self.pylogixPLC.Micro800 = False
		self.multiread = True`

And my reading code looks like this:

`if self.multiread:
	connected = False
	try:
		values = self.pylogixPLC.Read(["bool01", "int01", "float01", "string01"])
		connected = True
	except Exception as err:
		print("ERROR [" + self.pylogixPLC.IPAddress + "]: " + str(err))
		if "Multi-read failed" in str(err):
			self.multiread = False
		elif str(err) == "timed out":
			self.connected = False

if not self.multiread or not values:
	# Read tags individually`

Whenever I run the code, multi-read fails and my code falls back to individual reading.

dmroeder, 2019-02-06:
I just did a quick test and it does in fact work.  My test looked like this:

from eip import PLC
tags = ['bool01', 'int01', 'float01', 'string01']

p = PLC()
p.IPAddress = '192.168.1.9'
values = p.Read(tags)
print(values)

Now when you say your string is 82 characters, I assume you mean the default STRING data type, not a custom created string with 82 characters right?

Lucasrsv1, 2019-02-06:
Exactly. The default size is 82 characters, but it actually have something about 20 characters written.

dmroeder, 2019-02-07:
Are you running the latest pylogix code form github?  Back in June, I had a few problems reading multiple tags of mixed data types.

I ran your code exactly (from your other opened issue) and it works fine.

Lucasrsv1, 2019-02-07:
Yes, I'm running the latest code, and the last commit shown by `git log` is from Jan 31st 2019, which matches with github's commits page.
I'm wondering what are the odds of been some misconfiguration on the PLC...

dmroeder, 2019-02-07:
Since you say that you can read the tags individually, I don't think there is any kind of configuration issue.  Could you capture this problem with wireshark and email it to me?

Lucasrsv1, 2019-02-07:
Sure thing, I will schedule a visit with the company that owns the CompactLogix so I can capture the problem with wireshark. As soon as I get it, I'll send it to you.
What is your email, please?

dmroeder, 2019-02-07:
Thanks,

dmroeder@gmail.com

We'll discuss your other issue once we work through this one, I think they are going to end up being related.

Lucasrsv1, 2019-02-07:
About the Micro8xx, do you have any documentation that shows that the PLC itself really doesn't support multi-reading?

Lucasrsv1, 2019-02-07:
OK, thank you for your help!

dmroeder, 2019-02-07:
The PLC is telling you that the multi-read service is not supported, not my code.  I just translate the error code into text.

Lucasrsv1, 2019-02-07:
Got it, that's all I needed to know. Thank you once again!

dmroeder, 2019-02-21:
Did you happen to ever get this working, or get any details?

Lucasrsv1, 2019-02-21:
No, the company that owns the PLC hasn't schedule a visit for me yet. I'm still waiting for theirs disponibility. But, probably, it will take place next month.

dmroeder, 2019-02-21:
Ok, sounds good, I didn't want to leave you hanging.

kodaman2, 2019-09-27:
Closing as per #38 details.

closed 2019-09-27 by kodaman2