dmroeder/pylogix #38: ControlLogix Usage

Lucasrsv1, 2019-02-06:
Hi, I'm having an issue with a PLC ControlLogix 5571. When I connect to the PLC and try to read some tags I get the message "Path destination unknown" for any tag that I try. But I've managed to connect to the PLC and read those same tags using another driver in C# created by Allen Bradly. I'm also using the same configurations of IP and Slot.

Is there any step that I'm missing for ControlLogix usage?

Any help would be appreciated.

dmroeder, 2019-02-06:
There isn't anything special you need to do with CLX other than specify the slot of the PLC if it is not in slot zero.  Can you share your code and describe a little about this L71?  Like what is the ethernet modules address, what slot the L71 is in, etc.

Lucasrsv1, 2019-02-06:
Sure, my code looks like this:

```
from pylogix.eip import PLC
class PLCMonitor:
	def __init__ (self):
		self.pylogixPLC = PLC()
		self.pylogixPLC.IPAddress = "192.168.1.35"
		self.pylogixPLC.ProcessorSlot = 0
		self.pylogixPLC.Micro800 = False
		self.multiread = True

	def Read (self):
		if self.multiread:
			connected = False
			try:
				values = self.pylogixPLC.Read(["bool01", "int01", "float01", "string01"])
				connected = True
			except Exception as err:
				print("ERROR [" + self.pylogixPLC.IPAddress + "]: " + str(err))
				if "Multi-read failed" in str(err):
					self.multiread = False
				elif str(err) == "timed out":
					self.connected = False

		if not self.multiread or not values:
			# Read tags individually
```

I've also tried to change the slot to 1 and 2, but it didn't work.
When I connected using my other C# driver I used this configuration:

```
{
	"ip": "192.168.1.35"
	"port": 44818,
	"slot": 0
}
```

Using both drivers I tried to read these 4 tags: ["bool01", "int01", "float01", "string01"]. The C# driver gave me the right values, but this pylogix gave me the message "Path destination unknown" for all tags.

More information about the PLC itself, like its ethernet modules, I will be able to give you tomorrow, because a friend of mine is the guy who really knows about the PLC and its details.

yannickasselin, 2019-02-07:
I am very curious about this C# driver created by Allen Bradley. What is this driver?

Lucasrsv1, 2019-02-07:
Actually, I said it wrongly, the driver was developed by AdvancedHMI and you can find it [here](https://www.advancedhmi.com/index.php?main_page=product_info&cPath=3&products_id=14).

Lucasrsv1, 2019-02-07:
About the ControlLogix 5571, my friend said that the ethernet module is 1756-EN2TR and the processor slot is zero.

kodaman2, 2019-03-15:
If you ask me I think you are over thinking it, not too mention many programming mistakes. If the code posted is all you got, is never going to do anything at all. You have PLCMonitor class with init and a Read function, and that's it. Nowhere have you instantiated PLCMonitor class, and nowhere have you called Read which by the way is already implemented so not sure what you are trying to accomplish here.  And that's assuming you are not getting import errors. In other words this PLCMonitor class is useless, as you already have the PLC class.

Based on the code you provided I suggest you learn a bit more about how classes work. Nevertheless pylogix is really easy to use.

ReadMe and [examples](https://github.com/dmroeder/pylogix/tree/master/pylogix/examples) should get you started please read the documentation:

```
from pylogix import PLC
with comm as PLC()
    comm.IPAddress = '192.168.1.35'
    value = comm.Read('bool01')
    print(value)
```
If the code above or any from the example gives you an issue, please post errors.

Lucasrsv1, 2019-03-17:
Hi @kodaman2 , actually I posted just a small piece of my code just to show the context of the reading process. I have other classes using the PLCMonitor and a REST API to receive commands and instantiate and manipulate the PLCMonitor, which by the way has five more functions.

I've read the documentation and used the exemples to get started. My code is working for Micro8xx and CompactLogix when I read tags individually. So this issue is related to the use of this driver to communicate with a ControlLogix, which is not woking for me.

PS: I know how classes work, and pylogix is in fact awesome and really easy to use ;)

evaldes2015, 2019-03-17:
I have a test program that worked against a CompactLogix 1769-L16ER-BB1B. After seeing Lucas's post I tried it against a ControlLogix 1756-L71 and I didn't see any issues reading or writing to it. 
Lucas - are you sure you have the correct slot?

kodaman2, 2019-03-17:
@Lucasrsv1 controllogix, and compactlogix use same etherip protocol. Please post full code, or repo link to be able to help you out more otherwise we are just shooting in the dark. What I would do if I was you, is step trace debug. You'll be able to see where is your code failing, but this is definitely not a driver issue. I use this driver for both compactlogix, and controllogix.

TastySpaceApple, 2019-09-13:
Thank you all for this lean and efficient code. I have the same problem described here - I am unable to use the example code to read tags. I wonder if there are any pointers you can give me. Accessing the CLX PLC Tag from C# with AdvancedHDMI works, but using this wonderful piece of python makes the PLC return the unintelligible "Path destination unknown" error. From your experience, is there any difference you can think of in the way EIP Packets differ between the two open source libraries that could explain this?

kodaman2, 2019-09-13:
@TastySpaceApple Please start a new issue, and post your code.

TastySpaceApple, 2019-09-13:
> @TastySpaceApple Please start a new issue, and post your code.
Hello kodaman. Thank you very much for your reply. 
(1) The code is exactly the 5 line example code. It is basically also the code the OP was referring to, neither of us added anything substantial.
(2) I am not telling you how to run the repository but as it is the same issue, it would be better left on the same thread. 
(3) I see you have the attitude of "there can't be a problem, because it works on my machine" - please avoid it, and refrain from answering when you do not have an answer.

TastySpaceApple, 2019-09-13:
Hello kodaman. Thank you very much for your reply.
(1) The code is exactly the 5 line example code. It is basically also the code the OP was referring to, neither of us added anything substantial.
(2) I am not telling you how to run the repository but as it is the same issue, it would be better left on the same thread.
(3) I see you have the attitude of "there can't be a problem, because it works on my machine" - please avoid it, and refrain from answering when you do not have an answer.

kodaman2, 2019-09-13:
Asking for code isn't anything unusual. The code on the repo works. I asked to see code because usually people don't know how to write the tags. In order for us to help you have to give information as in any other repo.

- Post your code
- Tell us your setup, which slot is the plc
- Can you ping the plc?
- are you reading controller tags or program tags? 

Again we aren't psychic people.

Lucasrsv1, 2019-09-13:
After some tests I can ensure that my problem isn't connection configuration such as slot number. Even though I really want to share more of the code, I do not have permission to do so. But regarding the use of this library, there isn't much more to share.
My code is still fully functional when communicating with a Micro8xx PLC and this library helped me a lot, so I am really grateful!
In case it might help someone that is having this same issue related to ControlLogix, I've managed to use [this](https://github.com/plcpeople/nodepccc) PCCC library to communicate with ControlLogix.

dmroeder, 2019-09-13:
> Thank you all for this lean and efficient code. I have the same problem described here - I am unable to use the example code to read tags. I wonder if there are any pointers you can give me. [...]

It's strange that you guys are having trouble with CLX, I use it all the time with those controllers.  As far as AHMI, I'm very familiar with it and the author, the packets would be essentially identical with the exception of a few parameters possibly being different.

The "unintelligible" error code is a CIP error code, the PLC returns it if the tag cannot be found.  Typically, the tag name being read does not exist in the controller (typo) or you are trying to read a tag that actually exists in the scope of the program rather than the controller (not global).

I would like to get to the bottom of why you guys are having trouble with CLX, I'm not sure what I can do though.  There is no reason why it would not work with ControlLogix, as I'm typing this, I have a 1756-L61 and a 1756-EN2T reading/writing just fine. 

Can you get online with your CLX, create a new tag, say “SillyDint” of the DINT type, in the controller scope, then try to read it with pylogix?  Being online and creating the tag will guarantee that you are connected to the PLC you think you are, you know the tag name in fact exists in the PLC you think you are connecting to and it exists in the scope that you think it should.  I realize that you have successfully gotten AHMI to work, and you may be shaking your head at my suggestion, keep in mind that I don’t know anything about your hardware or how you are connecting.  Maybe you have that CLX sitting next to you on a desk, maybe you are in a factory with 100 controllers on a network.  I’ve helped people many times only to find out that they weren’t actually connecting to the PLC they thought they were.  Or in one case, where one PLC was working but the other wasn’t, turns out the tag name hand an upper case “i” in one PLC and a lower case “L” in the other, looks the same but obviously isn’t.  

If there is anything that you can’t share publically, you can always email me.

kodaman2, 2019-09-14:
@Lucasrsv1 and @TastySpaceApple Do us a favor and use my compiled utility https://github.com/kodaman2/Data_Preserve , and that will tell you anything you need to know since it logs everything.

Download released [latest](https://github.com/kodaman2/Data_Preserve/releases/tag/1.2.0), on the Settings.ini adjust accordingly, and add configuration file as explained on the README. I would recommend to add just a few tags to begin with. Any questions on my utility, open an issue on that repo or send me an email. And let us know if you were able to get it working, once is done, all data is saved in another file with _save postfix.

kodaman2, 2019-09-14:
Deleted last comment just got an even brighter idea, instead of using my datapreserve tool, I will create another one specifically for testing pylogix, and guide you to testing through prompts. Stay tuned!

kodaman2, 2019-09-16:
@Lucasrsv1 and/or @TastySpaceApple we're trying to figure out if there are some systems that are not compatible with pylogix, https://github.com/kodaman2/pylogix-tester/releases/latest, I wrote a utility for testing and this will ensure if you can read tags or not. If you can't please post log results.

Just download exe, and run, alternately you can run the py script after intalling pylogix, ping3 with pip. Hopefully this will help us get to the bottom of this issue.

Lucasrsv1, 2019-09-27:
Hi @kodaman2 , thank you for this utility, I used it today to read tags from a ControlLogix and it worked properly! Then, I used it to track down where on my code I was doing something wrong and I found a bug that was keeping the property Micro800 from being false. That was the problem, Micro800 was always true and that led to the "Path destination unknown" error while reading from a ControlLogix.

@TastySpaceApple since you're having the same error, I suggest you check out if you're setting the property Micro800 correctly.

Thank you all for the support, the library is amazing, and my problem was solved.

kodaman2, 2019-09-27:
@Lucasrsv1 You are very welcome, glad to hear you got it up and running. Closing issue for now.

@TastySpaceApple if you are still having problems after testing with [pylogix-tester](https://github.com/kodaman2/pylogix-tester/releases/latest) open a new issue with the following items:

- Code
- Py version
- pylogix version
- PLC Setup, controller model, slot
- OS System details
- Errors if any

closed 2019-09-27 by kodaman2

sabirzgaoua, 2020-05-07:
I could not Install pylogix, i did command "pip install pylogix" and I get this error: (see error attached)
[image]

kodaman2, 2020-05-07:
@sabirzgaoua Hi, please open a new issue. 

When you do open an issue answer the questions there. Also give your pip version.

```
pip --version
```