dmroeder/pylogix #41: How do I write to an array of short int in the PLC ?

alexvto, 2019-02-15:
Hello,

I need to write to an array of short integers (2 bytes) in the PLC.
I am defining my array like this:

 self.__B40: [int] = []

However, when setting bit 15 of one of the list elements the pylogix library throws this error:

CRITICAL  EXCEPTION:short format requires (-32768) <= number <= 32767 in plc_write_tag_array(<pylogix_eip.PLC object at 0x04309B10>,B40[0],([1, -65520, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 5207, 0, 5269, 0, 5273, 0, 4251, 0, 4253, 0, 4254, 0, 4255, 0, 4258, 0, 4259, 0, 4260, 0, 4261, 0, 4262, 0, 4263, 0, 5290, 0, 5292, 0, 5295, 0, 4279, 0, 4297, 0, 4096, 0, 4104, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 560, 8192, 568, 8192, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],))

Any ideas on how to deal with this ?
Thank you !!

dmroeder, 2019-02-15:
I think one way to deal with this is to use ctypes.  I believe python is treating your integer (short) as unsigned when in Rockwell land (for the most parts), integers are signed.  Try this and see the result:

```
from ctypes import c_short

x = 0
x = x | (1 << 15)
print x
print c_short(x).value
```

dmroeder, 2019-02-21:
Did this help your situation?

alexvto, 2019-02-21:
Yes it did!
Thank you.

I also tried an array("h").tolist() but that didn't work.

closed 2019-02-21 by dmroeder