dmroeder/pylogix #43: Writing to a read-only tag exception

kodaman2, 2019-03-13:
Making an issue just for record purposes, as we've already talked about it offline.

**Issue:**
The problem was found sort of by accident, I have two configuration files with tags in it, and I can save their online values, and later on I would attempt to load (write to plc). One tag was giving me problem, and Dustin found out it was indeed a read only tag. The configuration files probably shouldn't have read only tags, but I don't make those configuration files. Hence the problem.

- When writing to a tag that doesn't exist a try/except works as expected.
- When writing to a tag that exists, but is read-only program raises an exception with a traceback.

A few ideas, custom errors for the above two Errors, and whatever other ones.
```
... except (TagNotFoundError, ReadOnlyError, OtherError):
...     pass
```
[https://docs.python.org/3/tutorial/errors.html](url) User defined exceptions 8.5, I don't know if custom errors are needed or if you can just use the existing cip codes in eip.py somehow.

dmroeder, 2019-03-13:
I didn't really want to go down the path of raising custom errors because there are quite a few.  I made changes to raise ValueError more consistently.  Your code will work fine with these changes since you were checking for ValueError.

kodaman2, 2019-03-14:
Thanks for the fix, will test sometime today. Looks good, CIP Status code, plus the string should be more than enough.

dmroeder, 2019-03-14:
Even though ConHoldTime is set to Read/Write, the instance of UDT_TD_CfgData is set to ReadOnly.  That is, Cfg the first member of UDT_TD_UNIT, which is of the type UDT_TD_CfgData and it is set to ReadOnly.

kodaman2, 2019-03-14:
Ah ok that makes sense nested structure, and the parent is read only. So confusing lol.

closed 2019-03-17 by kodaman2