dmroeder/pylogix #44: read array of boolean

enriba, 2019-03-22:
I have an array of booleans of length 127.
why the function read ("arrayName", 127) gives me error?

I can only read one bit at a time with the Read function ("arrayName [index]")

evaldes2015, 2019-03-22:
I just did a test read of 128 elements from a BOOL array. It worked fine.

Are you sure your array length is correct in the PLC? Can you send an image of the controller tag configuration?

enriba, 2019-03-22:
[image]

evaldes2015, 2019-03-22:
print(comm.Read('test9', 128))

works just fine reading:
[image]

Don't know what to say. Double check your spelling?

enriba, 2019-03-22:
I'm a idiot, the function read() work very well.
I use pylogix in a flask restful api, the error is in my flask code

closed 2019-03-26 by dmroeder