dmroeder/pylogix #45: reading a tag of type UDT

snowwindwaverider, 2019-04-01:
I see pylogix is parsing the UDT members when retrieving the tag list which is awesome.  I don't seem to be able to read a UDT. CPU is 1756-L75 and firmware version is 32. 
I have a test program with the following controller scoped tags:
[image]

using this to get the tag list and read each tag:
`from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.197.4'
    tags = comm.GetTagList()
    
    for t in tags:     
        print("tag: ", t.TagName, ". dt:", t.DataType,  ". size:", t.Size)
        if (t.DataType.strip() != ""):
            print("value: ", comm.Read(t.TagName))`

I get the following output. (I have added a bunch of prints in eip.py)
`t:  b'p\x00^\x00\x01\x00g\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xa1\x00\x04\x00\x01\x00\x00 \xb1\x00J\x00\x05\x00\xcc\x00\x00\x00\x00\x00\xca\x00\x00\x00\x00\x00\x00\x00\xc4\x00\x04\x00\x00\x00my_UDT;nHAEA\x00udt_member1_real\x00udt_member2_dint\x00\x00\xc0\xcdC>'
size:  16
p:  b'\x00\x00\xca\x00\x00\x00\x00\x00\x00\x00\xc4\x00\x04\x00\x00\x00my_UDT;nHAEA\x00udt_member1_real\x00udt_member2_dint\x00\x00\xc0\xcdC>'
p_length:  68
going to take the last  16  bytes of p as memberBytes
memberBytes:  b'my_UDT;nHAEA\x00udt_member1_real\x00udt_member2_dint\x00\x00\xc0\xcdC>'
members:  [b'my_UDT;nHAEA', b'udt_member1_real', b'udt_member2_dint', b'', b'\xc0\xcdC>']
tempplate key: my_UDT
t:  b'p\x00b\x00\x01\x00g\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\xa1\x00\x04\x00\x01\x00\x00 \xb1\x00N\x00\x06\x00\xcc\x00\x00\x00\x00\x00\xc4\x00\x00\x00\x00\x00\x00\x00\xc4\x00\x04\x00\x00\x00my_UDT2;nEAEA\x00udt2_member1_dint1\x00udt2_member1_dint2\x00#\xec\xfa\x9c'
size:  16
p:  b'\x00\x00\xc4\x00\x00\x00\x00\x00\x00\x00\xc4\x00\x04\x00\x00\x00my_UDT2;nEAEA\x00udt2_member1_dint1\x00udt2_member1_dint2\x00#\xec\xfa\x9c'
p_length:  72
going to take the last  16  bytes of p as memberBytes
memberBytes:  b'my_UDT2;nEAEA\x00udt2_member1_dint1\x00udt2_member1_dint2\x00#\xec\xfa\x9c'
members:  [b'my_UDT2;nEAEA', b'udt2_member1_dint1', b'udt2_member1_dint2', b'#\xec\xfa\x9c']
tempplate key: my_UDT2
tag:  Program:MainProgram . dt:  . size: 0
tag:  my_dint . dt: DINT . size: 0
value:  8765
tag:  my_UDT_tag_instance . dt: my_UDT . size: 0
line 22, in <module>
    print("value: ", comm.Read(t.TagName))

  File "..\pylogix\eip.py", line 90, in Read
    return _readTag(self, tag, count, datatype)

  File "..\pylogix\eip.py", line 232, in _readTag
    return _parseReply(self, tag, elements, retData)

  File "..\pylogix\eip.py", line 1349, in _parseReply
    vals = _getReplyValues(self, tag, elements, data)

  File "..\pylogix\eip.py", line 1382, in _getReplyValues
    vals.append(str(s.decode('utf-8')))

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 2: invalid start byte
`
It appears to be seeing the datatype as 160 and thinking that is string.  If I had zero values in the UDT members it would sometimes print out random characters instead of dumping.

I have attached a wireshark pcap. 
[read_UDT_tag.pcap.zip](https://github.com/dmroeder/pylogix/files/3031744/read_UDT_tag.pcap.zip)

dmroeder, 2019-04-05:
Sorry for the long delay, I've been away from a computer for a bit.

First, there has never really been any intention to read the entire contents of a UDT.  I recently added the ability to get the UDT name with GetTagList() but didn't really plan on taking that much farther.

B) A STRING is a "struct" just like a TIMER, COUNTER or UDT.  Essentially anything but the base data types is a "struct" (160).  STRING though has an identifier that tells me that although it is a struct, it happens to be the standard STRING type so I know how to parse it.  If you try to read a UDT, TIMER, custom string type, etc, there isn't currently a way to tell the difference.

3) I could make a change so that when you try to read a UDT, it will just return the array of bytes and you'll have to understand how it is packed and unpack it yourself.  I have proposed this to a few other users but nobody tends to respond when I suggest it.  It's probably the way it should be anyway.

Hopefully I didn't get too far off course there about the string/struct stuff, it's been a while since I've had to dig into that so it's possible that that part of my memory got swapped out with some other nonsense :)

snowwindwaverider, 2019-04-12:
In the end the way for me to go was to parse the program l5x file so I know the structure of the UDTs and then read the members one by one. I ran in to problems with the Rockwell Library Of Process Objects (PAX) AOI defined types by trying to read them since some of the local tags have external access set to None. 

Thanks for your efforts in making pylogix! it is very handy to be able to pull the data out of the PLC without setting up an HMI software.

mos89, 2019-05-10:
hi is it possible to read a tag with a custom UDT and it will return the array of byte which can be cast to retrieve the data ?

dmroeder, 2019-05-10:
Hello mos89, it is possible to read the values of a UDT with a few caveats.  The reply needs to fit into 1 packet, so large UDT's probably won't work.  It's up to you to parse the reply.  In order to do it, you need to know the exact structure of the UDT, the order that the values will be coming in and the number of bytes of each value.

Regarding the idea of it fitting in 1 packet, I recently added the large forward open, so using ConnectionSize, you can increase the packet size to 4000 bytes.  Of course, your processor has to support that feature.

Example 40_read_timer.py is an example of reading a structure.  The same idea will apply to any UDT.  If you give me details of the UDT (maybe the export of it), I might be able to help you if you need it.

mos89, 2019-05-13:
hi thanks for your reply
here is the details of my UDT: see attached file
My tag is CW_BufferMes is an array of UDT :
My tag is a pretty large structure:if you can give me some indication to read tha data of  CW_BufferMes[1,0].Entete.Chantier for exemple this will be very helpful for me
thanks

dmroeder, 2019-05-16:
I did not see an attachment of your UDT.  What I think most users do when dealing with UDT's it to export your UDT to a L5X/L5K, then write a script to get the makeup of the UDT, which then can be used to read the tags.  The more complex the UDT, the harder this gets.  I'm working on an example of doing this which I'll push to the project as I get time.

mos89, 2019-05-17:
> I did not see an attachment of your UDT. What I think most users do when dealing with UDT's it to export your UDT to a L5X/L5K, then write a script to get the makeup of the UDT, which then can be [...]

Hi
i try a new link for the attachment
[https://wetransfer.com/downloads/e36ed8a1558130fbbc3cbbf16c7cd51920190517130318/25547cd4547245f08ec5404966903a9920190517130318/6862f6](url)

dmroeder, 2019-05-17:
LOL, that is a giant UDT.  I can't do much with a word document though.

First of all, it doesn't show me the entire structure.  Second, for me to write something to parse it, I would need the L5X export, even if your word document showed the entire structure of your ~30kb UDT, I'm not going to sit here and recreate it myself.  

If you can get me the actual L5X export of the entire UDT, depending on how much work it will be, I may be able to help you out.  Otherwise, you might just have to settle with an example of parsing a L5X and work it out for your UDT.  If there is something about your UDT that you don't want to share publicly, feel free to email it directly to me (see setup.py).

closed 2019-05-25 by dmroeder

shripalpanchal, 2023-04-28:
Below is sample code to read UDT. UDT contain  follwoing
1) Array of string.
2) Array of Real
3) Array of Dint

https://github.com/shripalpanchal/GP-Pylogix-Demo-v0-Test/blob/master/Udt_Class.py