dmroeder/pylogix #46: Speeding up reads

pr0ximo, 2019-04-12:
Hello dmroeder , 

First and foremost, thanks so much for making this module, it is much appreciated!

I'm currently reading about 100 Boolean per second, and I'd like to read faster.  I have a list of 440 tags in a variable called 'final_list', and I'm looping through each individual tag and executing a read command using the Pylogix module.  This is my code:

[image]

Looping and individual reads might be slowing it down, but it seems we need to because of the 500 byte limit you mentioned in the 05_read_multiple_tags.py example.  I've also tried following the fast read example, 09_multi_read_faster.py, but I can't seem to get it to work.  We've tried putting the data into tuples and nested arrays and neither worked.  Here is a picture with the tuples example: 

[image]

I'd like to be able to read faster, about 1000 tags per second.  Have you ever been able to read this fast with PyLogix?  If so, do you have any recommendations for speeding this up? I've been thinking about using Cython to speed up execution time, what are your thoughts?

Thanks,

proximo

dmroeder, 2019-04-12:
Can you explain what you mean by "doesn't work" when you try to do what is in your second picture?  I tried exactly that and it reads just fine.

Whether you could actually get 1000 reads per second reading individual tags will depend on your controller.  If it isn't the newer architecture controllers (5380/5580) with the gigabit ports, you won't be able to get that performance.

Utilizing lists of tags with read will get you better performance since you are packing more data per packet.  Of course you have to watch how many tags you attempt to read at a time (as you mentioned).

Providing the data type with the list of tags (tuple/list of list) will get you even better performance since it will pack more tags in a single packet as well as eliminate retrieving the data type of each tag.

For the best performance, map everything into an array of DINT's (for example) and read the entire array.

dmroeder, 2019-04-12:
I should also add you should move your reads into the "with" indent.
Also, don't print the results

kodaman2, 2019-05-12:
Even if your controller can handle it, and your network setup I wouldn't recommend reading that many tags per second, you can overload the comms % usage and bogged down communications for important stuff like HMI communications and controls logic. Go on a browser enter controller IP, and go under tasks to see if there is any high percentage on communications. I've actually seen systems with L73 where the cpu can't handle any more comms, and doing something so high performance like you are looking for can have negative effects on the system.

closed 2019-09-08 by dmroeder

kyle-github, 2019-09-08:
Oops, missed this the first time around.

What @kodaman2  said.   Reading that many individual tags is going to kill your PLC if you try to do it one read at a time.   I am not sure about pylogix, but I have done it with my library as a stress test.   That uses a lot of internal async handling with multiple partial requests in flight and absolutely requires multiple request packing and large packets (AB allows you to negotiate up to 4000-byte packets).   I've done up to 1000 DINT tags at a time, but it is really hammering the PLC!   Not recommended.

Using boolean arrays is almost always an anti-pattern in AB logic.   Reading them is a pain and the code in the PLC is a lot slower.  Use single bits in DINTs.

If you are trying to read that many tags individually, I strongly suggest copying them to a DINT array in the PLC itself and reading the whole array as a single tag.   Be aware that the network IO and the PLC scan are not synchronized with respect to each other.   So if you are writing these bits all out in a row in some PLC program, your tag read can get a partially updated array.   We use sentinel values at the beginning and end of each such array to determine if we got a clean read.   It breaks far more often than you would think.