dmroeder/pylogix #47: Returning tagValue and DataType with Read()

JJW152, 2019-04-16:
Great library.

One possible improvement I have that you might consider is the option to return the DataType along with the tag value when read() is called.

Like other users I'm sure, I am reading a giant list of tags that are all explicitly defined by the end user in an Excel sheet. They could be nested tags, UDTs, or array subscripts, so it is not possible to call GetTagList() and return an array that contains all the tags I need.

To improve speed, I would still like to read the tags with the data type explicitly defined if I have that information.

The idea here is that if I don't have the datatype, I can read the tag without it, cache the datatype on return, and then use it to improve speed the next time I read that tag.

I've created an external function to accomplish this, but I think it would be neat if it was a built-in feature.

Again, great work!

evaldes2015, 2019-04-16:
It kinda already does. pylogix builds a dictionary of tags as you read them. It uses the dictionary to speed up subsequent operations with that tag. You have easy access to the list.

Try something like this:

c = PLC()
c.IPAddress = '192.168.32.28'
c.ProcessorSlot = 0

tag = "test6"

v = c.Read(tag)
tagtype = c.KnownTags[tag][0]
tagtypestr = c.CIPTypes[tagtype][1]

print(f"{tag}:{v}:{tagtype}:{tagtypestr}")

kodaman2, 2019-05-12:
Hi JJW152, evaldes2015 had the right idea, is actually much simpler without accessing the dictionary have a look at the available class variables for class LgxTag for reference:
[eip.py lines 183 to 193](https://github.com/dmroeder/pylogix/blob/master/pylogix/eip.py#L183-L193)

    class LgxTag:
	    def __init__(self):
		    self.TagName = ''
		    self.InstanceID = 0x00
		    self.SymbolType = 0x00
		    self.DataTypeValue = 0x00
		    self.DataType = ''
		    self.Array = 0x00
		    self.Struct = 0x00
		    self.Size = 0x00

Dustin has done a great job with providing examples for most tasks with this library:
[Example 23](https://github.com/dmroeder/pylogix/blob/master/pylogix/examples/23_get_all_tags.py)

    from pylogix import PLC
    
    with PLC() as comm:
	    comm.IPAddress = '192.168.1.9'
	    tags = comm.GetTagList()
	    for t in tags:
		    print(t.TagName, t.DataType)

Hope this answers your question.

closed 2019-09-27 by kodaman2