dmroeder/pylogix #48: Reading Local Variable from Micro820

mglowinski93, 2019-04-21:


dmroeder, 2019-04-22:
Unfortunately, there is no way to read local variables, Rockwell doesn't provide a way.  You can only read global variables.

mglowinski93, 2019-04-22:
At least it is clear now. Thanks for reply.

closed 2019-04-22 by mglowinski93

evaldes2015, 2019-04-22:
Is this issue specific to the Micro820? Because I can read program locals on an CompactLogix.

dmroeder, 2019-04-22:
This is specific to the Micro820 and 850.

mglowinski93, 2019-04-23:
I can ask Rockwell Directly (i'm cooperating with them very often), if you tell me what to ask about.

bashiryousufy, 2022-10-05:
@mglowinski93 Did you find a way to read local variable?

mglowinski93, 2022-10-05:
> @mglowinski93 Did you find a way to read local variable?

No, i didn't.

bashiryousufy, 2022-10-05:
> > @mglowinski93 Did you find a way to read local variable?
> 
> No, i didn't.

Did you find any alternatives?

So basically I want to be able to read counter data and also be able to reset the counter using this package. Do you think its possible?

Note: Using Mirco820 PLC

mglowinski93, 2022-10-05:
> > > @mglowinski93 Did you find a way to read local variable?
> > 
> > 
> > No, i didn't.
> 
> Did you find any alternatives?
> 
> So basically I want to be able to read counter data and also be able to reset the counter using this package. Do you think its possible?
> 
> Note: Using Mirco820 PLC

No, I did not.
I simply mapped local variables to global ones, it was enough in my case.

bashiryousufy, 2022-10-05:
> No, I did not. I simply mapped local variables to global ones, it was enough in my case.

Any tips on how to achieve the mapping?

mglowinski93, 2022-10-05:
> Any tips on how to achieve the mapping?

I simply used the `COPY` instruction.