dmroeder/pylogix #50: Can this library work with Omron PLC?

arorarl, 2019-04-30:
I have an Omron NX1P PLC which supports CIP communication. Here is the link to the manual of the controller I am using which states that. 

https://www.google.com/url?sa=t&source=web&rct=j&url=https://assets.omron.eu/downloads/manual/en/v2/sysmac_nx_csg320_users_manual_en.pdf&ved=2ahUKEwjl7NC_5PjhAhVFiHAKHf2iCX8QFjAAegQIBhAC&usg=AOvVaw06v95OrpSrpuhi-QEevZqf

Also I have tried using cpppo python library for CIP connection. The github project link is https://github.com/pjkundert/cpppo

I have been able to successfully identify the PLC on the network by using broadcast option using the command

```python -m cpppo.server.enip.client --udp --broadcast --list-identity -a 192.168.1.255```

This brings me to my question, can I use this module to read my PLC which I believe supports CIP and have evidence for backing it up? Or I am completely wrong and lost :(

If this library supports my PLC, can you help me in someway to create tags for/from the variables that the MAIN program is currently using.

Please help and thanks in advance for your time :)

dmroeder, 2019-05-01:
I would say that it probably won't be able to do anything with that PLC.  It is for Rockwell PLC's only.  Even though the Omron supports CIP, I believe it is intended to do exactly what this library does, read/write tags from a Rockwell PLC, not the other way around.

arorarl, 2019-05-01:
Thanks for your time to answer me. However could you please give me some pointers as to what I should be changing in the code to make it work with my Omron PLC. What is the part of the code which if I modify will allow me to send and receive explicit unconnected messages to my PLC?

Once again I really appreciate your time.

dmroeder, 2019-05-01:
It sounds like I could be mistaken and that it may be possible.  The only problem is that I have no experience with Omron PLC's and have no access to them, so I'm not sure how I can help.  I don't know what configuration (if any) would need to be done on the Omron end.

What happens when you try to read something?

xinthose, 2019-05-07:
@arorarl Just use modbus TCP/IP.

jaredchesnut, 2019-05-08:
you definitely can use it on an nx1p, I'm doing it on an NJ, I'll have to
drum up the code changes later. it involved changing the cip protocols,
manufacturer, that kinda thing.

dmroeder, 2019-05-10:
arorarl, did you give it a shot?

jaredchestnut, thanks for posting.  I'd be curious to see the changes you made.

closed 2020-03-12 by kodaman2