dmroeder/pylogix #51: Values with GetTagList

DCogent, 2019-05-15:
First off great job on the this. Made it fairly painless to connect to a PLC. 
My issue though: When I use your GetTagList, I can pull in name and type no problem, but no value.
Im trying to grab all the tags names and values and write them to a libre file.  To do this I have tried combining your "multiple log example' with your "get all tags example" by trying to inject the tags = with the whole t.TagName list. It always fail when It tries to read the values. 

Any help on this would be great appreciated.

dmroeder, 2019-05-16:
You are most likely getting error when using the GetTagList result directly because there are probably a few items that aren’t really tags, like program names.  I should look into better filtering the reply.  You can try something like this:

```
for t in tags:
    if t.DataType:
        value = comm.Read(t.TagName)
```

Some of those nonsense things in the “tag list” won’t have a data type, so that should filter them out.

You need to keep in mind, that anything which is a structure (UDT) will not be parsed into something useful, it will be a list (array) of bytes.  Although Pylogix does understand that something is a UDT, it does not understand the structure of it.  This would be up to you to handle.  There are two main ways that I think people can handle reading UDT’s:

1)	Write a class that matches the structure of the UDT and unpack the array of bytes in that class.  My example 40_read_timer.py is a simple example of this.  This requires knowing in detail how data structures are packed.
2)	Write a parser that will parse L5X/L5K exports of the UDT from your program so that you know the makeup of the UDT ahead of time. When you see a tag of that UDT, you can use your parser to know how it is made up and read the individual elements.  The advantage of this is that you don’t really have to know how data is going to be packed into a byte array from the PLC.

DCogent, 2019-05-16:
Thank you very much for the quick reply. I will try this.

KHDOUDI, 2019-05-22:
Hello 
Thank you very much for this high quality work that is now serving lot of users around the world.

I have small issue with GetTagList (), after starting the program it take maybe a while (2 seconds) trying to collect the tags and after it says that connexion lost.

I know that i have some thousands of tags that should be returned and maybe the PLC close the connection before Pylogix get the full list ?

Any solution for this please ?

dmroeder, 2019-05-23:
Hmm, I haven't experienced that before.

Can you send me a wireshark of it happening?
Can you send me your exact code?

KHDOUDI, 2019-05-23:
Hello, Thank you for your feedback,

i used the same code as the examples you provide with documentation:

`import sys
sys.path.append('..')

from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.1.9'  # Here i put the IP of my PLC
    tags = comm.GetTagList()
    
    for t in tags:
        print(t.TagName, t.DataType)`

I will try to capture a wireshark next time and send it to you.
I tested other examples for simple tag read and it's working fine.

kodaman2, 2019-07-03:
@KHDOUDI Hey did you get it sorted out? Please send screenshot of this connection lost error or paste exact error here. Run below ping command, and try to run GetTagList().

```
ping 192.168.1.9 -t
```

Another question is what PLC model do you have, it sounds like you might have a different slot for the PLC. Can you confirm your CPU is on slot 0?

closed 2019-09-27 by kodaman2