dmroeder/pylogix #53: Exception: Forward Open Failed

PriteshKanani, 2019-05-31:
I have been running this code to read tag value of Micro850 PLC.

    import time
    from pylogix import PLC
    comm = PLC()
    comm.Micro800 = True
    comm.IPAddress = '192.168.1.129'
    while True:
        print(comm.Read('_IO_EM_DI_00'))
        print(comm.Read('_IO_EM_DO_00'))
        time.sleep(1)

After running it from pycharm IDE, It throws below error.

    C:\ProgramData\Anaconda3\python.exe D:/Learning/rockwell_test.py
    Traceback (most recent call last):
      File "D:/Learning/rockwell_test.py", line 10, in <module>
        print(comm.Read('_IO_EM_DI_00'))
      File "C:\ProgramData\Anaconda3\lib\site-packages\pylogix\eip.py", line 91, in Read
        return _readTag(self, tag, count, datatype)
      File "C:\ProgramData\Anaconda3\lib\site-packages\pylogix\eip.py", line 201, in _readTag
        if not _connect(self): return None
      File "C:\ProgramData\Anaconda3\lib\site-packages\pylogix\eip.py", line 801, in _connect
        raise Exception("Forward Open Failed")
    Exception: Forward Open Failed
    
    Process finished with exit code 1

However, When I use default Python IDLE, then program runs fine. One thing to note is that when I tested both interpreter with test code:

    import pylogix
    print(pylogix.__version__)

Both of them works fine. I can not determine where is the fault. It would be really great help if someone can help me out here. Thank you.

dmroeder, 2019-05-31:
The short answer is that I broke the Micro8xx series when I added the large forward open.  There are some default connection parameters that aren't compatible with that series.  I don't ever use that PLC, so I didn't test the latest on it.

The reason it is working in one interpreter is most likely that PyCharm is using python3 and IDLE is using python2 , you probably have two slightly different code bases installed between python2 and python3.

I know what is wrong, I'll get it fixed soon, hopefully tonight.

dmroeder, 2019-06-01:
The problem should be fixed with the latest commit.

PriteshKanani, 2019-06-02:
I have verified. The issue has been resolved. Thank you for the efforts @dmroeder . It has been really helpful.

closed 2019-06-02 by PriteshKanani

dmroeder, 2019-06-02:
I'm glad it has been helpful, thanks for the feedback!