dmroeder/pylogix #55: Reading UDT member names

ciderglider, 2019-06-12:
I’ve been using your program to communicate with an Allen Bradley 1769-L24ER-QBFC1B device. I’ve been looking at how the Allen Bradley device reports its UDTs. I see there’s a function _getUDT() that attempts this. 

I’ve declared a UDT on the Allen Bradley device that has three members, a SINT, a DINT and a REAL, so nothing too exotic. When I run the get_program_tags.py example, the Template Object Definition Size is extracted from the Get Attribute List response and uses this to populate the “bytes of data to read” field in the Template Read request (multiplied by 4, minus 23, then padded for 32-bit alignment). The Template Read response has 9 bytes at the end that I can’t make sense of; they don’t correspond with anything that’s in Allen Bradley “Logix 5000 Controllers Data Access” document. Are these 9 bytes some reserved data being sent by Allen Bradley, or is the populate the “bytes of data to read” field in the Template Read request being set up incorrectly in this situation?

This issue doesn't seem to happen with Predefined Data Types.

dmroeder, 2019-06-19:
Hey ciderglider.  It's been some time since I have dug into that portion, at a glance, it appears that I'm requesting too many bytes and it is returning some extra garbage at the end of the member names.  I'll have to dig into this again to get it properly sorted out, which I'll do soon...

ciderglider, 2019-06-24:
Thanks for the response.

dmroeder, 2019-07-11:
I got a little side tracked... Honestly, I'm not sure what those extra bytes are.  I think I was asking for too many bytes by padding the request, I'm can't remember why I settled on that.  The Controller Data Access document doesn't include that.  Even without that, there are still a few extra bytes being sent by the PLC.

ciderglider, 2019-07-11:
No worries. I agree with what you've said: the padding isn't mentioned in the Rockwell Controller Data Access document, but that even if that padding is taken out, extra bytes are still read. I have a feeling that the number of extra bytes correlates with the number of members the UDT has; maybe there are 3 bytes per member?

Also, your code works fine with Predefined Data Types.

closed 2020-03-12 by kodaman2