dmroeder/pylogix #58: pylogix with AB micrologix series PLCs

sharanrajc, 2019-06-29:
Hi,
I am trying to use pylogix driver with micrologix series of Allen Bradley PLCs but I'm unable to connect and read tags. Has anyone tried it before? Specifically I'm working with Micrologix 1400 B series PLC.

dmroeder, 2019-06-29:
This will not work with the Micrologix series PLC's.  Only ControlLogix, CompactLogix and Micro8xx.

sharanrajc, 2019-07-01:
@dmroeder thanks for the reply. pylogix has the most extensive apis for interacting with PLC that I have come across. What can be done to make it work with micrologix series PLC's? I have a Micrologix 1400 PLC at office and I want to make pylogix work with it, how can I contribute?

dmroeder, 2019-07-01:
I don't own one, nor do I use them at my current job.  If I end up owning one some day, I may think about supporting it, but I currently have no plans.

Have you ever checked out AdvancedHMI, or pycomm?

sharanrajc, 2019-07-04:
I have tried pycomm and it definitely works with ML1400 for individual register read and write . What I miss in pycomm is features such as discover and readtaglist which is available in pylogix.

kyle-github, 2019-07-04:
There are two main classes of PLC from AB: those that are CIP native such as ControlLogix, CompactLogix and Micro8xx, and those that are PCCC encapsulated in CIP such as PLC/5, SLC 500 and MicroLogix.   The latter is something of a pain as you are really dealing with two different protocols with the PCCC (almost DF1) packet encapsulated in the outer CIP packet (which is encapsulated in the EIP packet...).

Support for PCCC-based PLCs is a serious annoyance.   I tried just supporting PLC/5 commands and found that on MicroLogix, some of the addressing modes just silently fail or fail with an error when they work just fine on a PLC/5.   I ended up having to split up my library support into four different categories:

1. ControlLogix and CompactLogix,
2. Micro8xx (doesn't support things like request bundling and a few other things),
3. PLC/5,
4. SLC 500, MicroLogix

Each type of PLC seems to have its own little quirks and differences in the supported commands in the protocol and in how it returns values (PLC/5 returns strings padded but MicroLogix does not seem to do that, Micr8xx strings are returned with a one byte string length counter and ControlLogix returns them with a 4-byte counter).

TL;DR: once you go down the path of breaking this down by PLC type, it doesn't stop :-(

evaldes2015, 2019-07-04:
All of the PLCs that require PCCC are EOL aren't they? 
Perhaps sharanrajc should try a library that supports PCCC. There are a few out there.

dmroeder, 2019-07-04:
Thanks for the info Kyle.  Supporting SLC/Micro has been a rabbit hole that I have had no interest in going down.  Your post confirms it :)

sharanrajc, 2019-07-09:
Thanks @kyle-github and @dmroeder for clarifying the underlying complexities. @evaldes2015 I would definitely lookout for PCCC supporting libraries. Thanks for the input.

kyle-github, 2019-07-09:
@evaldes2015, yes, almost all (all?) of the PCCC-based PLCs are EOL, but there are soooooooooooo many of them still out there.

dmroeder, 2019-07-09:
Yeah even though they are EOL, and as Kyle points out, there are a ton of them, the'll still be around for a long time.  We still get requests to work on 5TI's and PLC2's.

I think it would be fun to experiment with, however I only have so much bandwidth. So I prefer to just stick with the Logix platform.

ottowayi, 2019-07-09:
@sharanrajc if you're using Python 3, the tag list feature was added [here](https://github.com/bpaterni/pycomm/tree/pycomm3).  I used some of @dmroeder's work to get that and the extended forward open working in pycomm.  I also have a repo called pycomm3 that I'm working on too that I plan on adding a bunch more features to, but haven't gotten around to it.

evaldes2015, 2019-07-09:
@sharanrajc I have used TuxEIP in the past. It is a C library. It's kinda hard to find and the version on Github has bugs that I fixed and tried to push back but could never get it to work. I really don't like Git. 
If you want to take a look at my version, it can be found here: https://github.com/leicht/TuxPLC/issues/11#issuecomment-323602563
I posed a zip and a tar.gz with the code. FYI, this code has been running at multiple sites for several years.

closed 2019-07-18 by dmroeder

MEPA77, 2023-01-03:
Just Use Pycomm 3 from https://github.com/ottowayi/pycomm3.git
quick example on reading 10 integers from micrologix 1100 over ethernet:

from pycomm3 import SLCDriver
with SLCDriver('192.168.1.107',) as plc:    #set the SLC/Micrologix driver and IP address to use
         N= plc.read('N7:0{10}')            # read 10 integers starting at N7:0
print (N[1])

6lv1-blr, 2023-08-08:
> Just Use Pycomm 3 from https://github.com/ottowayi/pycomm3.git quick example on reading 10 integers from micrologix 1100 over ethernet:
>
> from pycomm3 import SLCDriver with [...]

works perfectly for my need, a connection before done with an OCX asabtcp replace by a call to python via python4delphi.
For a micrologix 1400
thanks.