dmroeder/pylogix #59: Using PL to Send Images to PLC

KalebSixto, 2019-07-11:
Is there a way to use Pylogix to send an image or multiple images to the PLC?

evaldes2015, 2019-07-11:
You could write it as a DINT, INT or SINT array. 
Why do you want to do this?

dmroeder, 2019-07-11:
Could you please clarify what you are trying to do?  I don't understand what you mean by "send an image".

KalebSixto, 2019-07-12:
I am processing a changing graph using matplotlib and seaborn. I want to send the images of the updating graphs to a control logix plc.

evaldes2015, 2019-07-12:
What would the PLC do with it?

KalebSixto, 2019-07-12:
The plc would output the images to a tool or monitor.

evaldes2015, 2019-07-12:
You're using the PLC to transfer the image? Surely there are better mechanisms than this.

KalebSixto, 2019-07-12:
I most certainly agree, but where i work they want to use the PLC. I'm not certain why. I am merely following instructions.

evaldes2015, 2019-07-12:
The max array size in ControlLogix is 2MB. Will your image fit?
Another issue is performance. Assuming it takes 6ms to transfer 100 array elements to the PLC, it would take 30 seconds to write a 500KB image. This is best case.

KalebSixto, 2019-07-12:
I will have to check into the size of the images im outputting, however I am certain at least one of the images is too large. You mentioned before that there might be a better way. What would you recommend?

evaldes2015, 2019-07-12:
It depends. Where are the images coming from? Where are they going? Can they be saved as files? It all depends on the details. Can you FTP them from one system to another? Can you write them to a shared folder?

KalebSixto, 2019-07-12:
The images are coming from a computer in python. They are being outputted to machine monitors inside of labs. The rest of the questions I will have to look into as I am uncertain due to my lack of experience with matplotlib.

evaldes2015, 2019-07-12:
You should be able to write image files in various formats from Python.

KalebSixto, 2019-07-12:
I got permission to not use the plc for the image processing. Thanks for the help!

closed 2019-07-18 by dmroeder