dmroeder/pylogix #61: Read array of strings

10062244, 2019-08-06:
Hello,
Im encountering issues when I try to read an array of strings. Even when I try to read a single element of the array I get an error. Is this action not possible for this data type?

Thanks,

dmroeder, 2019-08-06:
You should be able to read arrays of STRING data types, as long as they are the standard STRING type and not a custom STRING type.

10062244, 2019-08-06:
The issue turned out to be the external access for that specific tag was set to none in the PLC. With that changed, it works great. Thanks for the awesome lib and the quick response.

closed 2019-08-06 by 10062244

dmroeder, 2019-08-06:
Ahh, good call.  I often forget about external access since we don't ever restrict anything.