dmroeder/pylogix #62: Error Using Emulate ControlLogix - Exception: Forward Open Failed

rafaelmoraes33, 2019-08-14:
I have a problem.

I have a Studio 5000 Emulate and created a Slot with ControlLogix.

How much I try to connect is giving this error: Exception: Forward Open Failed

If I perform the Discovery function it usually appears my device and IP.

There is some specific configuration.

Attached is the image of how it is configured.

My code:

import pylogix
from pylogix import PLC
pylogixPLC = PLC()

pylogixPLC.IPAddress = "192.168.0.22"
pylogixPLC.ProcessorSlot = 2
pylogixPLC.Micro800 = False

pylogixPLC.Read('tag')

Thank you

[image]

dmroeder, 2019-08-14:
Emulate does not support Ethernet communication so you cannot use this library to communicate with it. You can use SoftLogix if you have it.

rafaelmoraes33, 2019-08-14:
Many thanks for the return.

But if I use hardware from a CompactLogix would that work?

Thanks

dmroeder, 2019-08-14:
It will work with CompactLogix over Ethernet.

rafaelmoraes33, 2019-08-14:
It worked super well, thank you very much.

Another thing, when it is Strig type there is some way to convert as it does not come with normal characters.

Thanks

dmroeder, 2019-08-15:
Are you reading the standard string type, or a custom length string? The standard string should be returned exactly as you see it in your program.

closed 2019-09-08 by dmroeder

dmroeder, 2021-08-19:
I have confirmed with a user that pylogix will communicate with Emulate by setting ConnectionSize to 504.

TheFern2, 2021-08-19:
> I have confirmed with a user that pylogix will communicate with Emulate by setting ConnectionSize to 504.

I can work on adding this to the documentation sometime in the morning, do you want to perhaps do a helper param or function to active emulate mode?

dmroeder, 2021-08-20:
I'm going to change the default fall back connection size to 504 bytes to make it compatible with emulate by default, I'll also add a not to the readme.

TheFern2, 2021-08-20:
I'll test in the morning I got a vm with emulate I've been using for performance testing on ft linx ua. Unless you already have tested. This is huge for people that have no hardware to test until plc arrives.