dmroeder/pylogix #63: Forward Open Failed on 1756-ENET/B

damienmorrissey, 2019-08-27:
I'm not sure where this would fit, but I have discovered that, when using the 1756-ENET/B module, that the forward open will fail if the comm.ConnectionSize is greater than 508. I don't know for sure why, but I am guessing it might be MTU related?

ottowayi, 2019-08-27:
That is because the large connections sizes are only supported on EN2T and newer, as well as logix v20 and newer.  The old stuff is limited to 500 bytes.

damienmorrissey, 2019-08-27:
Thanks for that. Yeah- I figured that would be the case. Would it be worth
me adding a note to the readme file to that effect? I realise not many will
have such old gear, but there would be a few.

damienmorrissey, 2019-08-27:
> I will lower the default value so that you won't have to mess with it. I
> prefer that the default works with everything out of the box, users can
> increase it if their hardware/firmware supports it.
>
> I think the oldest hardware I tested with was a ENBT/A and a 1769-L32E.
> Strangely they support up to 511 bytes, which is why I made the change.

Ah that makes sense. I agree- defaults working with everything makes life
easier. Let me know if you want me to test anything.

damienmorrissey, 2019-08-28:
Hi dmroeder we just tested with the latest commit and it worked first go.

Thank you for making that change. It's actually for a dev system made of old parts so it wouldn't have been seen in production. I've not used one of these ENET cards in a long time either- they're very old.

Would you like me to close this? Not used Github before :).