dmroeder/pylogix #64: Can't read Timer on example 40 get a string stead

armandom84, 2019-08-29:
I was trying to use the example to read a timer and a UDT and I keep getting string as response instead of bytestream and I'm going crazy because I can't find what is happening. I think this happens because the current value is 0 in my timer, but can't see why.

Any help will be super appreciated.

dmroeder, 2019-08-29:
Can you share your exact code with me?  You can email it if you prefer.  I used the example 100% as is, it works fine.  There is no reason whey a ACC value of 0 would cause any problem, that is a perfectly legit number.

I would consider reading data structures an advanced concept.  If you use the Read method on a tag that is a data structure, it is going to return a byte stream that you are going to have to parse yourself.  This requires knowing exactly how the data is going to be packed into the byte stream. 

The fact that you are seeing a byte stream suggests that you are not unpacking the bytes out of the response like in the example.  I put the timer example together simply to show how you can read entire data structures, with a data type that is familiar with everyone.  I wouldn't waste the time using it in practice, I'd just make a list of tags and read them:

tags = ['TestTimer.PRE', 'TestTimer.ACC', 'TestTimer.EN', 'TestTimer.TT', 'TestTimer.DN']
response = comm.Read(tags)
print(response)

If you are not actually after the values of the entire structure, then just read the stuff you want.

armandom84, 2019-08-29:
Thank you for your super quick response.

I know that maybe is out of my level, but even going just to the example
you made I'm still confused about why it's failing.

This is my code; i just changed the address and the timer to read to one
that is already on my plc code.

'''
the following import is only necessary because eip.py is not in this
directory
'''
import sys
sys.path.append('..')
from pylogix import PLC
from struct import pack, unpack_from

'''
Reading a UDT will return the raw byte data, it is up to you as the user
to understand how it is packed and how to unpack it.  Looking at the data
type in L5K format can be helpful.

This example is reading the entire structure of a timer in 1 read.
For a timer, bytes 2-5 contain the .EN, .TT packed at the end of the word.
Bytes 6-9 contain the preset and bytes 10-13 contain the acc value
'''
class Timer(object):

    def __init__(self, data):

        self.PRE = unpack_from('<i', data, 6)[0]
        self.ACC = unpack_from('<i', data, 10)[0]
        bits = unpack_from('<i', data, 2)[0]
        self.EN = get_bit(bits, 31)
        self.TT = get_bit(bits, 30)
        self.DN = get_bit(bits, 29)

def get_bit(value, bit_number):
    '''
    Returns the specific bit of a word
    '''
    mask = 1 << bit_number
    if (value & mask):
        return True
    else:
        return False

with PLC() as comm:
    comm.IPAddress = '192.168.1.2'
    ret = comm.Read('T[0]')
    t = Timer(ret)
    print(t.PRE, t.ACC, t.EN, t.TT, t.DN)

the error i get is :

Traceback (most recent call last):
  File
"C:\Users\amartinez\Desktop\python\pylogix-master\pylogix\examples\40_read_timer.py",
line 42, in <module>
    t = Timer(ret)
  File
"C:\Users\amartinez\Desktop\python\pylogix-master\pylogix\examples\40_read_timer.py",
line 22, in __init__
    self.PRE = unpack_from('<i', data, 6)[0]
TypeError: a bytes-like object is required, not 'str'

I can read directly acc , preset if I look directly for them. The problem
seems to be that i don't get a bytestream but a str

Any help will be appreciated, I really like what you are doing!

dmroeder, 2019-08-29:
So you have an array defined called "T" that is the data type TIMER[n]?  Your are 100% certain that "T[0]" is not a STRING type?
Can you tell me which PLC you are connecting to and the firmware revision?

armandom84, 2019-08-29:
Yes I have an array T, its a TIMER[300] array with all of the timers used
on the program. This is a controller level array and 100% sure that is not
an String.

I tried also with a differente UDT (not inside an array of itself) (this
was a UDT made by myself to represent the specific information been shared
with a specific temperature controller); that udt is composed by an INT a
secondary UDT that contains a set of 5 bools (not in an array) and another
INT; when I try to read it ; if I check the type of the data coming back
type(x) I still get it back as an str. (I tried it just to see if with a
different UDT i will get a different response).

Right now I'm using a 1768-L45 CompactLogix5345 Controller running on
revision 20.18 .

dmroeder, 2019-08-29:
Strange, I cannot recreate the behavior, unless I use the example to read a STRING type.  I don't have a L43 to test with, I wouldn't expect there to be a L43 specific problem.  However I've seen some unexpected and odd things.

Could I convince you to capture the behavior with Wireshark and send me the packets?

armandom84, 2019-08-29:
Here is what I saved from Wireshark don't have much experience with it but
hope it is enough.

dmroeder, 2019-08-30:
Can you do the following in your python interpreter and let me know what it outputs?

```
>>> import pylogix
>>> pylogix.__version__
```

armandom84, 2019-08-30:
0.3.1  is the version that is showing

dmroeder, 2019-08-31:
That is what I thought.  I bet you are one commit away from the change I made to return the raw bytes when the data type is a struct.  This change was to allow the ability to read the values of a UDT.  

Grab the latest code and it should work.

armandom84, 2019-08-31:
Thank you so much for this help. Now it is working. You are amazing.

dmroeder, 2019-08-31:
Great, I'm glad it's working!

closed 2019-08-31 by dmroeder