dmroeder/pylogix #66: ProgramList and String no working with Micro800

punisher97, 2019-09-19:
Hi, i working your wonderful package, i couldn't get de program list, only get global tags, i'm working with Micro820, By chance you have tried with this PLC? thanks

**PDTA**: I'm working in my custom package written in Node.js, please you see and say me your opinion, i port partialy your python code to Node.js, thanks 

I hope it is useful to someone: [node-logix]( https://www.npmjs.com/package/node-logix)

punisher97, 2019-09-19:
I found response from AdvancedHMI,  other library, thanks,

> Thanks for the update. The Micro800 seems to be quite limited with using CIP symbolic addressing. So far, these are limitations I have found:
> 
> - Early versions of firmware did not support symbolic addressing at all
> - Only Global variables can be read
> - Multi-service request doesn't work which can make refreshing a lot of values very slow
> 
> and now adding to the list:
> 
> - Variables in UDFs not accessible
[source](https://www.advancedhmi.com/forum/index.php?topic=1531.0)

closed 2019-09-19 by punisher97

dmroeder, 2019-09-20:
Yeah, the Micro series is pretty feature limited, I guess you can expect that for the prices.  Funny thing about them is they supported unsigned types and 64bit types before the more expensive Logix platform did.  Go figure.