dmroeder/pylogix #67: Atomic writes for UDTs

nicksanford, 2019-09-20:
Thank you so much `dmroeder` for the amazing library!

I'm trying to be able to atomically update multiple fields of a UDT in a single packet. I am currently able to update members of my UDT, but I need to break each member update into it's own write call.  If I try to write to the struct tag itself I get an error (below). I would love to learn how to do this if it is currently supported by pylogix, or potentially submit a PR to add support for it if you think it would be both possible and useful.

Say I have a UDT with 2 DINTS, I would like to be able to update them both without needing to make 2 Write calls:

```
from pylogix import PLC
with PLC() as conn:
    conn.IPAddress = "10.0.0.1"
    udt_data = conn. Read("UDT_TAG") # => bytes for the entire UDT
    conn. Read("UDT_TAG.first_member") # => 1
    conn. Read("UDT_TAG.second_member") # => 2

    conn.Write("UDT_TAG.first_member", 3)
    conn.Write("UDT_TAG.second_member", 4)

    # now say I want to revert the tag back to before I did the two writes, I'd like to be able to do:
    conn.Write("UDT_TAG", udt_data)
```

This is the error when I try to write to "UDT_TAG" directly:
```

~/repo/plc_adapter/python/plc_client.py in write(plc_ip, tag_name, value)
     47     with PLC() as conn:
     48         conn.IPAddress = plc_ip
---> 49         conn.Write(tag_name, value)
     50
     51 def size_of_tag_request(tag_name):

~/repo/plc_adapter/python/venv/lib/python3.7/site-packages/pylogix/eip.py in Write(self, tag, value, datatype)
     96         the arguments, write a single tag, or write an array
     97         '''
---> 98         return _writeTag(self, tag, value, datatype)
     99
    100     def MultiRead(self, tags):

~/repo/plc_adapter/python/venv/lib/python3.7/site-packages/pylogix/eip.py in _writeTag(self, tag, value, dt)
    264             writeData.append(float(v))
    265         elif dataType == 160 or dataType == 218:
--> 266             writeData.append(MakeString(self, v))
    267         else:
    268             writeData.append(int(v))

~/repo/plc_adapter/python/venv/lib/python3.7/site-packages/pylogix/eip.py in MakeString(self, string)
   1622         work.append(ord(char))
   1623     for char in string:
-> 1624         work.append(ord(char))
   1625     if not self.Micro800:
   1626         for x in range(len(string), 84):

TypeError: ord() expected string of length 1, but int found
```

Please let me know if there is any more info I can provide which would be useful and thank you very much for your time.

kodaman2, 2019-09-22:
Hi Nick unfortunately what you are trying to do with UDT's is not supported. Both read/write UDT operations you have to explicilty write the UDT_Name.member. Reading the raw bytes isn't very useful by itself unless you know how to unpack the data. [Explained here](https://github.com/dmroeder/pylogix/blob/01e7f84b5b937418a76abe4d74943c8b677124cd/pylogix/examples/40_read_timer.py)

You can find similar discussions on UDT's here, addressed by Dustin:

https://github.com/dmroeder/pylogix/issues/17#issuecomment-487719123
https://github.com/dmroeder/pylogix/issues/45#issuecomment-480440027

nicksanford, 2019-09-23:
Hi Fernando!

Thank you very much for the info. 

I am able to pack / unpack the UDTs by referencing L5X files. What I am not able to do is update the entire UDT (potentially all members of that UDT) in a single TCP call.

That said, I now understand from https://stackoverflow.com/questions/54695471/are-cip-services-for-reading-and-writing-plc-tags-on-ab-logix-5000-controllers that even if I could use pylogix to update an entire UDT in a single tcp cal, I still don't have a guarantee that the PLC will apply updates to each member of the struct atomically. So to your point, it looks like what I am trying to do is simply not supported.

closed 2019-09-23 by nicksanford