dmroeder/pylogix #68: GetTagList() has AttributeError: 'Response' has no attribute, 'value'

ambersnow, 2019-09-23:
Hi, trying to use GetTagList() via your examples in the repo but I am not having any luck getting it to work.

My code:
```python
from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.1.200'
    tags = comm.GetTagList()
    for t in tags:
	    print(t.TagName, t.DataType)
```

The trackback:
```python
Traceback (most recent call last):
  File "pylogixtest.py", line 5, in <module>
    tags = comm.GetTagList()
  File "C:\Users\Admin\Test\venv\lib\site-packages\pylogix\eip.py", line 127, in GetTagList
    updated_list = self._getUDT(tag_list.value)
AttributeError: 'Response' object has no attribute 'value'
```

Is there anything I am doing wrong or?

Seabee1971, 2019-09-23:
I ran into a similar problem. Can you ping the 192.168.1.200?

ambersnow, 2019-09-23:
Yep, sure can. I have 2 PLCs, one on .200 and another on .171 both giving me this same AttributeError.

evaldes2015, 2019-09-23:
Are your PLCs in slot 0?

ambersnow, 2019-09-23:
Yes — I even specified the `comm.ProcessorSlot = 0` but still no difference as well.

evaldes2015, 2019-09-23:
Are the tags in the PLC configured for external access?

ambersnow, 2019-09-23:
Yeah, they are all configured external access Read/Write. Pylogix has no problem reading/writing specific tags. Just no luck with `GetTagList()`

evaldes2015, 2019-09-23:
Is there something weird in your tag list? Some odd character in a name perhaps? I'm guessing...

ambersnow, 2019-09-23:
As far as I can tell, no. It’s all alphanumeric and underscores. Tags are all uppercase, if that makes a difference.

ambersnow, 2019-09-23:
After some digging, I found what was going on. The 2nd call in `GetTagList` was looking for `tag_list.value` when the `self._getTagList` response is `tag_list.Value`. Changing line 127 from:
`updated_list = self._getUDT(tag_list.value)`
To
`updated_list = self._getUDT(tag_list.Value)`
Is what gave me a list of all the tags and tag types :)

kodaman2, 2019-09-24:
@ambersnow  the latest version 0.4.0 has a slight change in the way the data is returned.

[23_get_all_tags example](https://github.com/dmroeder/pylogix/blob/master/pylogix/examples/23_get_all_tags.py)

Basically you have one minor error:

```
from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = '192.168.1.200'
    tags = comm.GetTagList()
    for t in tags:
	    print(t.TagName, t.DataType)
```

You need to add Value in the forloop like this:

```
for t in tags.Value:
...
```

In addition to that typo as you've said there is some typos in the latest commit. Good job, submitting the PR.

ambersnow, 2019-09-24:
No worries, this library is amazing compared to some others I've (tried to) used so anything to help contribute and make it better. :)

kodaman2, 2019-09-24:
Awesome I've added a small comment on the PR.

closed 2019-09-25 by kodaman2