dmroeder/pylogix #75: The response of GetProgramTagList didn't include DataType?

su600, 2019-09-26:
Hello!

When I using   
    **`tags=comm.GetProgramTagList("Program:MainProgram")`**
This function **didn't return DataType of program tags** as `tags= comm.GetTagList()`, I want get both TagName and Datatype of **program tags**, could you please check this?

Thank you

kodaman2, 2019-09-26:
Hey su600 please ensure you have downloaded the very latest version 0.4.4 as that function had a small bug  in the previous release. 

Then try this:
```
tags = comm.GetProgramTagList("Program:MainProgram")
if tags.Value is None:
    print(tags.Status)
else:
    for t in tags.Value:
        print(t.TagName)
```

su600, 2019-09-26:
My version is 0.4.3 now, I'll update it. 

But, anything about DataType?  Can I get DataType by GetProgramTagList?
Thanks

kodaman2, 2019-09-26:
ok, yeah looks like a bug, even in 0.4.4, I'll try to fix it within a few days. And ensure the unittest covers both TagName and DataType.

Thanks for reporting it!

kodaman2, 2019-09-26:
Please download repo from this commit, and let me know if it worked. Once I get the confirmation I can merge to master.

https://github.com/kodaman2/pylogix/tree/program-list-patch-02

kodaman2, 2019-09-26:
This issue is fixed, but I'll leave issue open until confirmation.

su600, 2019-09-27:
Yeah, It works well. 
(Something wrong I can't upload a screenshot）

`Program:MainProgram.l BOOL`
`>>> pylogix.__version__`
`'0.4.5'`

Thank you，this  issue is fixed.

kodaman2, 2019-09-27:
Awesome, you're welcome!

closed 2019-09-27 by kodaman2