dmroeder/pylogix #77: pylogix.eip.response object

pr0ximo, 2019-09-26:
Hello, 

I am trying to use the pyloglix library in a virtual environment, and whenever I read data from the PLC I get: 

<pylogix.ip.Response object at 0x02F7> 

I am not sure why it only does this in a virtual environment but not in my normal global environment.

kodaman2, 2019-09-26:
Hey pr0ximo the latest version returns Response object in order to avoid runtime exceptions.

Run the following to check pylogix version:
```
pip list
```

We're currently working out some small bugs, but for the next week or so keep checking for updated versions. Latest is 0.4.4. 

Try the following in the virtual environment:
```
from pylogix import PLC
with comm as PLC()
    comm.IPAddress = '192.168.1.9'
    ret = comm.Read('MyTagName')
    print(ret.TagName, ret.Value, ret.
```

A quick sanity check to ensure you have a value and not a None object.

```
if ret.Value is None:
    print(ret.Status)
else:
    print(ret.Value)
```

If for whatever reason you enjoy using the old way without Response object, please read the readme for installation instructions.

ottowayi, 2019-09-26:
You should add __str__ and __repr__ to the response class so it prints usable information.  Best practices say you should be able to exec the repr of an object and get the equivalent object.
 
    def __repr__(self):
        return f'Response({self.TagName}, {self.Value}, {self.Status})'

    def __str__(self)
       return f'{self.TagName}, {self.Value}, {self.Status}'

Most IDEs will use the repr for debugging, but print will call str on the object.

pr0ximo, 2019-09-26:
thanks guys! 
Just wanted to make sure I wasn't going nuts.  Keep up the great work!

kodaman2, 2019-09-26:
@ottowayi Yes, that is a good suggestion. Do you want to submit PR patch?

kodaman2, 2019-09-26:
> thanks guys!
> Just wanted to make sure I wasn't going nuts. Keep up the great work!

For sure, it was discussed and obviously it was a major change. I didn't like the idea at first since I was dealing with the exceptions at first with try/catch, but after using the latest the code is cleaner by checking if `ret.Value is None` than having all those try/catches all over the place in the client code.

kodaman2, 2019-09-26:
Closing issue, but added @ottowayi suggestion to the todo list, but PRs are welcome too.

closed 2019-09-26 by kodaman2

th3dj, 2020-03-26:
So i spent about an hour on this problem. It seems that in Python 3 its looking for a lowercase instead of an uppercase.
`
#!/usr/bin/env python3
import sys
#sys.path.append('..')
import csv
from pylogix import PLC
plcip = '000.000.00.000'

with PLC() as comm:
    comm.IPAddress = plcip
    ret = comm.Read('Error_Flag')
    #print(ret.TagName, ret.Value, ret.Status)
    if ret.value is None:
        print(ret.Status)
    else:
        print(ret.Value)
    
comm.Close()
`
Doesnt work,
However,
`
#!/usr/bin/env python3
import sys
#sys.path.append('..')
import csv
from pylogix import PLC
plcip = '000.000.000.000'

with PLC() as comm:
    comm.IPAddress = plcip
    ret = comm.Read('Error_Flag')
    #print(ret.tagtame, ret.value, ret.status)
    if ret.value is None:
        print(ret.status)
    else:
        print(ret.value)
    
comm.Close()
`
seems to work. Thank you visual studio....

kodaman2, 2020-03-26:
> So i spent about an hour on this problem. It seems that in Python 3 its looking for a lowercase instead of an uppercase. I Googled around tried the print and got the object reference. i searched the message and it brought me here. I figure that if someone ended up here it will help to have this information.
> `
> Doesn't work,
> #!/usr/bin/env python3
> import sys
> #sys.path.append('..')
> import csv
> from pylogix import PLC
> plcip = '000.000.00.000'
> 
> with PLC() as comm:
> comm.IPAddress = plcip
> ret = comm.Read('Error_Flag')
> #print(ret.TagName, ret.Value, ret.Status)
> if ret.value is None:
> print(ret.Status)
> else:
> print(ret.Value)
> 
> comm.Close()
> 
> Does work
> #!/usr/bin/env python3
> import sys
> #sys.path.append('..')
> import csv
> from pylogix import PLC
> plcip = '000.000.000.000'
> 
> with PLC() as comm:
> comm.IPAddress = plcip
> ret = comm.Read('Error_Flag')
> #print(ret.tagtame, ret.value, ret.status)
> if ret.value is None:
> print(ret.status)
> else:
> print(ret.value)
> 
> comm.Close()
> `

Thanks @th3dj but this is the wrong information. Pylogix is all uppercase, the most likely scenario is that you install `pip install pylogix` which is a rogue pypi package which we do not support from this github repositorie. That guy fork this repo when the value was lowercase, then Dustin fixed it to ensure it was like all other variables in the library. Again uppercase is correct for this repo. What you'd like to do is: 

```
import pylogix
pylogix.__version__
```

If it does not match any of our releases then that's your issue:

https://github.com/dmroeder/pylogix/releases

Btw to install this repo with pip follow instructions on this repo readme.

dmroeder, 2020-03-26:
@kodaman2 is correct, you are not using the main project but a fork of the main project.  Also, if you are instantiating using a "with" statement, don't call .Close(), this is handled automatically.