dmroeder/pylogix #81: Hacktoberfest help wanted, pep8 issues

kodaman2, 2019-10-01:
Hey github peeps,

I've noticed there are a few pep8 issues on the main file `eip.py` that would be perfect for first time submitters, and help you stack a PR for hacktoberfest rep.

https://github.com/PyCQA/pycodestyle

If you are new to git, an awesome guide is here https://github.com/firstcontributions/first-contributions

Pycodestyle is what used to be known as pep8, you run that module against a py file and it tells you syntax errors. That is your mission to help this repo be free of pep8 issues. Can you help us?

For example this is a branch I am working on, and I have been really naughty:

```
py -3.7 -m pycodestyle --first pylogixTests.py
pylogixTests.py:3:64: W291 trailing whitespace
pylogixTests.py:24:1: E402 module level import not at top of file
pylogixTests.py:25:17: E261 at least two spaces before inline comment
pylogixTests.py:225:80: E501 line too long (80 > 79 characters)
```

GilSandler, 2019-10-01:
Hey, 
A first-time open source contributor here :)
Will try to fix the issue.

kodaman2, 2019-10-01:
Thanks @GilSandler any questions feel free to ask.

kodaman2, 2019-10-01:
@GilSandler  I think the only pep issue we can leave is the W291 trailing whitespace at the top of the file, but other than that is all game, if you want to fix 5 or so, and leave some for others to contribute but is up to you. You can fix all if you'd like.

GilSandler, 2019-10-01:
@kodaman2 Actually already fixed everything except for the `E501 line too long` errors.
According to PEP8, lines have to be less than 80 characters long (Had a look, and I think splitting the "long" lines will make the code less readable. Moreover, PyCharm's standard, for example, is less than 120 characters per line, which the code adheres to).
Wanted to ask you what you think is best? Should I split the lines or not?

GilSandler, 2019-10-01:
Created a PR for the issue: #82

closed 2019-10-01 by kodaman2